COVESA/vsomeip #3: Reboot detection vs. TR_SOMEIP_00256

tolstoi911, 2017-04-06:
Looking at service_discovery_impl::is_reboot(), I don't think the reboot detection is lining up with TR_SOMEIP_00256, which states:

> The information for the reboot flag and the Session ID shall be
kept for multicast and unicast separately as well as for every sender-receiver-relation
(i.e. source address and destination address).

As I can see, the reboot flag is not kept separately for multicast and unicast (the sessionID is kept separately, but not the reboot flag).
This leads to issues once multicast and unicast reboot flags have different values.

GenivivSOMEIPmaintainer, 2017-04-11:
This will be addressed with vsomeip 2.7.0.

GenivivSOMEIPmaintainer, 2017-06-21:
vsomeip 2.7.0 is released.

closed 2017-06-21 by GenivivSOMEIPmaintainer