COVESA/vsomeip #4: FTBS using gcc 7

jmglo, 2017-06-08:
vsomeip fails to build using gcc7 (fedora26).
I've tried vsomeip 2.5.3 and 2.6.2

Build errors follows:

```
[ 11%] Building CXX object CMakeFiles/vsomeip.dir/implementation/endpoints/src/netlink_connector.cpp.o
In file included from /home/julo/Downloads/vsomeip-2.6.2/implementation/endpoints/src/netlink_connector.cpp:13:0:
/home/julo/Downloads/vsomeip-2.6.2/implementation/endpoints/src/../include/netlink_connector.hpp: In function ‘bool vsomeip::operator==(const vsomeip::nl_endpoint<Protocol>&, const vsomeip::nl_endpoint<Protocol>&)’:
/home/julo/Downloads/vsomeip-2.6.2/implementation/endpoints/src/../include/netlink_connector.hpp:101:28: error: no match for ‘operator==’ (operand types are ‘const sockaddr_nl’ and ‘const sockaddr_nl’)
         return e1.sockaddr == e2.sockaddr;
                ~~~~~~~~~~~~^~~~~~~~~~~~~~
/home/julo/Downloads/vsomeip-2.6.2/implementation/endpoints/src/../include/netlink_connector.hpp: In function ‘bool vsomeip::operator!=(const vsomeip::nl_endpoint<Protocol>&, const vsomeip::nl_endpoint<Protocol>&)’:
/home/julo/Downloads/vsomeip-2.6.2/implementation/endpoints/src/../include/netlink_connector.hpp:108:30: error: no match for ‘operator==’ (operand types are ‘const sockaddr_nl’ and ‘const sockaddr_nl’)
         return !(e1.sockaddr == e2.sockaddr);
                  ~~~~~~~~~~~~^~~~~~~~~~~~~~
/home/julo/Downloads/vsomeip-2.6.2/implementation/endpoints/src/../include/netlink_connector.hpp: In function ‘bool vsomeip::operator<(const vsomeip::nl_endpoint<Protocol>&, const vsomeip::nl_endpoint<Protocol>&)’:
/home/julo/Downloads/vsomeip-2.6.2/implementation/endpoints/src/../include/netlink_connector.hpp:115:28: error: no match for ‘operator<’ (operand types are ‘const sockaddr_nl’ and ‘const sockaddr_nl’)
         return e1.sockaddr < e2.sockaddr;
                ~~~~~~~~~~~~^~~~~~~~~~~~~
/home/julo/Downloads/vsomeip-2.6.2/implementation/endpoints/src/../include/netlink_connector.hpp: In function ‘bool vsomeip::operator>(const vsomeip::nl_endpoint<Protocol>&, const vsomeip::nl_endpoint<Protocol>&)’:
/home/julo/Downloads/vsomeip-2.6.2/implementation/endpoints/src/../include/netlink_connector.hpp:122:28: error: no match for ‘operator<’ (operand types are ‘const sockaddr_nl’ and ‘const sockaddr_nl’)
         return e2.sockaddr < e1.sockaddr;
                ~~~~~~~~~~~~^~~~~~~~~~~~~
make[2]: *** [CMakeFiles/vsomeip.dir/build.make:207: CMakeFiles/vsomeip.dir/implementation/endpoints/src/netlink_connector.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:241: CMakeFiles/vsomeip.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

GenivivSOMEIPmaintainer, 2017-06-19:
Please try this patch:

From 32089cb53561920062d378d2cb6109008ccf94e6 Mon Sep 17 00:00:00 2001
From: Lutz Bichler <Lutz.Bichler@bmw.de>
Date: Mon, 19 Jun 2017 15:44:49 +0200
Subject: [PATCH] Fix compilation with gcc7.

---
 .../endpoints/include/netlink_connector.hpp           | 19 +++++++++++++++----
 1 file changed, 15 insertions(+), 4 deletions(-)

diff --git a/implementation/endpoints/include/netlink_connector.hpp b/implementation/endpoints/include/netlink_connector.hpp
index db0cbc5f..a451566e 100644
--- a/implementation/endpoints/include/netlink_connector.hpp
+++ b/implementation/endpoints/include/netlink_connector.hpp
@@ -14,6 +14,7 @@
 
 #include <map>
 #include <mutex>
+#include <tuple>
 
 #include <boost/asio/ip/address.hpp>
 #include <boost/asio/basic_raw_socket.hpp>
@@ -98,28 +99,38 @@ public:
     friend bool operator==(const nl_endpoint<Protocol>& e1,
                const nl_endpoint<Protocol>& e2)
     {
-        return e1.sockaddr == e2.sockaddr;
+        return std::tie(e1.sockaddr.nl_family,
+                        e1.sockaddr.nl_groups,
+                        e1.sockaddr.nl_pid) ==
+               std::tie(e2.sockaddr.nl_family,
+                        e2.sockaddr.nl_groups,
+                        e2.sockaddr.nl_pid);
     }
 
     /// Compare two endpoints for inequality.
     friend bool operator!=(const nl_endpoint<Protocol>& e1,
                const nl_endpoint<Protocol>& e2)
     {
-        return !(e1.sockaddr == e2.sockaddr);
+        return !(e1 == e2);
     }
 
     /// Compare endpoints for ordering.
     friend bool operator<(const nl_endpoint<Protocol>& e1,
               const nl_endpoint<Protocol>& e2)
     {
-        return e1.sockaddr < e2.sockaddr;
+        return std::tie(e1.sockaddr.nl_family,
+                        e1.sockaddr.nl_groups,
+                        e1.sockaddr.nl_pid) <
+               std::tie(e2.sockaddr.nl_family,
+                        e2.sockaddr.nl_groups,
+                        e2.sockaddr.nl_pid);
     }
 
     /// Compare endpoints for ordering.
     friend bool operator>(const nl_endpoint<Protocol>& e1,
               const nl_endpoint<Protocol>& e2)
     {
-        return e2.sockaddr < e1.sockaddr;
+        return !(e2 > e1);
     }
 
     /// Compare endpoints for ordering.
-- 
2.13.1

GenivivSOMEIPmaintainer, 2017-06-21:
The next release (2.7.1) will support gcc7.

jmglo, 2017-06-26:
build works with patch, thx

closed 2017-09-18 by GenivivSOMEIPmaintainer

cornelizbasa, 2018-01-18:
I think there might be an error in the patch. I think the patch should be changed for `operator>` function as:
```
     friend bool operator>(const nl_endpoint<Protocol>& e1,
               const nl_endpoint<Protocol>& e2)
     {
-        return e2.sockaddr < e1.sockaddr;
+        return !(e1 < e2 || e1 == e2);
     }
```