COVESA/vsomeip #5: ARM build support  documents?

kumarasamy, 2017-07-11:
Hello ,

Please guide me how to build the vsomeip for ARM boards ?

GenivivSOMEIPmaintainer, 2017-09-18:
Building of vsomeip is documented in the vsomeip user guide. If you have a build environment with cmake and boost, building should work the same way it does on other platforms.

kumarasamy, 2017-09-25:
Thanks for your reply.But i am trying to port this to android platform for that reason i asked this question.

gunnarx, 2017-09-28:
Link to user guide please?  It's not in the README.

gunnarx, 2017-10-25:
Thanks.  (Can the issue be closed now?)

GenivivSOMEIPmaintainer, 2017-11-10:
@kumarasamy: This sound more like a problem with the cross(=) build environment. The cmake you use for building the ARM target should not find host components.

closed 2017-11-10 by GenivivSOMEIPmaintainer