COVESA/vsomeip #6: Register subscription status handler

NelutuOmat, 2017-07-20:
Is it possible to register a subscription status handler from CommonAPI/SomeIP?

In vsomeip, the function application_impl::register_subscription_status_handler registers
all subscriptions, but when ACK is received, the function application_impl::deliver_subscription_state is never reached.
The received _event id is always ANY_EVENT in the function application_impl::on_subscription_status, , which doesn't match the event ids saved in subscription_state_ map. So for all ACKs I get errors like:
[trace] 1343 application_impl::on_subscription_status: Received a subscription status without subscribe for 65/1/cc/ffff/error=0

tolstoi911, 2017-11-09:
Knock knock... anyone working on this?

jeremiah, 2017-11-09:
I think there is work being done on this. It is hard to tell because vsomeip doesn't really work in the open. There is a mailing list: https://lists.genivi.org/pipermail/genivi-ipc/ but even there discussions don't seem to be moving forward. I'll try and point this out to the maintainer(s) and get them to address the issues and mailing list topics.

GenivivSOMEIPmaintainer, 2018-05-23:
CommonAPI's Event::subscribe takes an optional errorListener that (despite its name) is used to register a status handler for the event.

closed 2018-05-23 by GenivivSOMEIPmaintainer