COVESA/vsomeip #9: Registering application: 3 failed. It is already registered!

ykhokhulya, 2017-09-21:
I'm working with CommonAPI IPC (vsomeip binding) and I've found strange behaviour in working of IPC.
At least such behaviour is not described in documentation (I looked only https://github.com/GENIVI/vsomeip/wiki/vsomeip-in-10-minutes). 
                                                           
If start more than 2 applications communicating via vsomeip; and after that stop
first started one (according to logs and code review it's HOST routing manager, others are PROXIES), then:
1. Communication between PROXIES will freeze untill new application is not started (Could you clarify is that OK?)
2. IPC communication will continue after starting of new application (will be started as HOST).
3. New HOST routing manager generates ids for new clients begining from 0. If start new application it gets already used client id.

At the 3rd step I have a lot such messages in HOST routing manager log:
```
  [2017-09-21 20:58:11.728455] [0x00007f9e2ffff700] [error] Registering application: 2 failed. It is already registered!
  [2017-09-21 20:58:11.728658] [0x00007f9e2f7fe700] [info]  REGISTERED_ACK(0002)
  [2017-09-21 20:58:12.828707] [0x00007f9e2f7fe700] [info]  Application/Client 0002 is registering.
  [2017-09-21 20:58:12.828772] [0x00007f9e2ffff700] [error] Registering application: 2 failed. It is already registered!
  [2017-09-21 20:58:12.829487] [0x00007f9e35d93700] [info]  REGISTERED_ACK(0002)
  [2017-09-21 20:58:13.929462] [0x00007f9e2f7fe700] [info]  Application/Client 0002 is registering.
  [2017-09-21 20:58:13.929886] [0x00007f9e2ffff700] [error] Registering application: 2 failed. It is already registered!
  [2017-09-21 20:58:13.930407] [0x00007f9e2f7fe700] [info]  REGISTERED_ACK(0002)
  [2017-09-21 20:58:15.030011] [0x00007f9e2f7fe700] [info]  Application/Client 0002 is registering.
  [2017-09-21 20:58:15.030079] [0x00007f9e2ffff700] [error] Registering application: 2 failed. It is already registered!

```

log of application started with conflicting client id:
```
  [2017-09-21 20:58:08.033699] [0x00007f57cb7fe700] [info] Application/Client 2 is registered.
  [2017-09-21 20:58:11.728563] [0x00007f57cb7fe700] [info] Application/Client 2 is registered.
  [2017-09-21 20:58:12.828908] [0x00007f57cb7fe700] [info] Application/Client 2 is registered.
  [2017-09-21 20:58:13.930035] [0x00007f57cb7fe700] [info] Application/Client 2 is registered.
  [2017-09-21 20:58:15.030237] [0x00007f57cb7fe700] [info] Application/Client 2 is registered.
  [2017-09-21 20:58:16.130970] [0x00007f57cb7fe700] [info] Application/Client 2 is registered.
  [2017-09-21 20:58:17.231244] [0x00007f57cb7fe700] [info] Application/Client 2 is registered.
  [2017-09-21 20:58:18.770072] [0x00007f57d72e3800] [info] Stopping vsomeip application "".
  [2017-09-21 20:58:18.770767] [0x00007f57cb7fe700] [info] Application/Client 2 is deregistered.
```

log of application with this client id:
```
  [2017-09-21 20:58:13.828746] [0x00007f28415f0700] [warning] Client 0x2 register timeout! : Restart route to stub!
  [2017-09-21 20:58:13.929072] [0x00007f28415f0700] [info]    Client 2 successfully connected to routing  ~> registering..
  [2017-09-21 20:58:14.929274] [0x00007f28415f0700] [warning] Client 0x2 register timeout! : Restart route to stub!
  [2017-09-21 20:58:15.029648] [0x00007f283b7fe700] [info]    Client 2 successfully connected to routing  ~> registering..
  [2017-09-21 20:58:16.029976] [0x00007f28415f0700] [warning] Client 0x2 register timeout! : Restart route to stub!
  [2017-09-21 20:58:16.130399] [0x00007f28415f0700] [info]    Client 2 successfully connected to routing  ~> registering..
  [2017-09-21 20:58:17.130571] [0x00007f28415f0700] [warning] Client 0x2 register timeout! : Restart route to stub!
  [2017-09-21 20:58:17.230908] [0x00007f283b7fe700] [info]    Client 2 successfully connected to routing  ~> registering..
  [2017-09-21 20:58:17.724841] [0x00007f284748d800] [info]    Stopping vsomeip application "".
  [2017-09-21 20:58:18.725383] [0x00007f283bfff700] [warning] 2 registering timeout on stop
  [2017-09-21 20:58:18.725963] [0x00007f28415f0700] [info]    utility::auto_configuration_exit: munmap succeeded.
```

Attached simple test applications enough for the issue reproducing.
[vsomeip-test.zip](https://github.com/GENIVI/vsomeip/files/1322376/vsomeip-test.zip)

GenivivSOMEIPmaintainer, 2018-05-23:
The general behavior is as expected. But the logs look different with 2.10.21 (I got no timeouts and no failed (re-)registrations).

closed 2018-05-23 by GenivivSOMEIPmaintainer

spiderman07, 2019-02-25:
Hi. What was the root cause for this? I have same bug in my application.

gunnarx, 2019-02-25:
Who are you asking for root cause @spiderman07 ?  I assume it is @ykhokhulya, but they never came back to confirm if they got their particular problem fixed or not.  
Because @GenivivSOMEIPmaintainer seems to say that it is expected behavior and not a bug.

I'd recommend you continue by starting with the provided examples and documentation and build out from there.  

Overall, if you're getting similar error messages then listen and think about what they are saying.
It sounds like two applications are registering themselves with the same application ID.  Are they, or are they not?  Study your config JSON files and keep trying to get it right.

spiderman07, 2019-02-25:
I was asking @ykhokhulya. But i think he got it sorted and never posted the answer. Thanks for feedback. I will try. Will post the result when i find something.

ykhokhulya, 2019-02-25:
The problem was:
First launched application performs IPC routing, it's so called HOST.
If the process of HOST terminated then IPC breaks: communication freezes, launching of new application leads to warnings in logs.

It seems VSomeip doesn't handle this case.

I've solved this issue in my environment by running vsomeipd deamon and explicitly specifying the routing via vsomeipd in /etc/vsomeip.json.

```
{
    "routing" : "vsomeipd"
}
```

gunnarx, 2019-02-25:
This is also how I've been successful before, by using an external vsomeipd daemon for routing.  Agree with the above - the communication stack will automatically start a thread to handle routing / registry for any program if nothing else is specified, but it seems more robust to have it separate from the app itself.

spiderman07, 2019-02-26:
Let's say i can't use vsomeip daemon for routing. Is there any other way to fix this? I mean what is the recovery for VSomeip? If VSomeip daemon crashes, do i need to reset the whole system? Cause if it is affected by this bug it might also fail on routing if it is restarted.

spiderman07, 2019-03-08:
I also managed to fix my issue. Nothing wrong with vsomeip. We are using boost sockets and some guy decided to do a change in boost code and, in the closing of the socket itself, remove the actual closing of the socket and just report everything works fine. Fixed his mistake and all is fine.