COVESA/vsomeip #10: symbol not found under alpine

Siddius, 2017-10-12:
I am trying to build vsomeip in a alpine:3.6 docker image.

first of all `BIG_ENDIAN` and `LITTLE_ENDIAN` is not defined.
so i changed StringEncoder.hpp so it will be set.

after this you can build it, but you get the following error : 
```
/ # ldd /usr/local/lib/libvsomeip-sd.so.2.7.0 
	ldd (0x564db5b5f000)
	libvsomeip.so.2 => /usr/local/lib/libvsomeip.so.2 (0x7f0afb5c5000)
	libboost_system-mt.so.1.62.0 => /usr/lib/libboost_system-mt.so.1.62.0 (0x7f0afb3c1000)
	libboost_log-mt.so.1.62.0 => /usr/lib/libboost_log-mt.so.1.62.0 (0x7f0afb0ed000)
	libstdc++.so.6 => /usr/lib/libstdc++.so.6 (0x7f0afad9b000)
	libgcc_s.so.1 => /usr/lib/libgcc_s.so.1 (0x7f0afab89000)
	libc.musl-x86_64.so.1 => ldd (0x564db5b5f000)
	libboost_thread-mt.so.1.62.0 => /usr/lib/libboost_thread-mt.so.1.62.0 (0x7f0afa961000)
	libboost_filesystem-mt.so.1.62.0 => /usr/lib/libboost_filesystem-mt.so.1.62.0 (0x7f0afa748000)
Error relocating /usr/local/lib/libvsomeip-sd.so.2.7.0: _ZN7vsomeip12payload_implC1Ev: symbol not found
```

jmglo, 2017-10-16:
Could be maybe solved adding 
vsomeip::payload_impl::payload_impl*;

to exportmap.gcc

Siddius, 2017-10-16:
Yep that was the mistake, works like a charm now. Thx

closed 2017-11-10 by GenivivSOMEIPmaintainer

jmglo, 2018-03-22:
Issue is closed could be resolved upstream ?

maiminhtien94, 2018-05-16:
Dear all when I add my code. I build it show error:
/home/root/request_respond/src/service-example.cpp:13:10: error: 'payload_impl' in namespace 'vsomeip' does not name a type
 vsomeip::payload_impl::payload_impl*;
          ^~~~~~~~~~~~
make[2]: *** [CMakeFiles/service-example.dir/build.make:63: CMakeFiles/service-example.dir/src/service-example.cpp.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:105: CMakeFiles/service-example.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
root@h3ulcb:~/request_respond/build#

gaetanprin, 2018-05-18:
hi guys i'm facing the same issue but no luck with export map... Any ideas ? It works well on ubuntu laptop but when i build for raspberry using yocto build system it compile well but at runtime I have the issue..

gaetanprin, 2018-05-18:
when i "`arm-nm libvsomeip.so` i'm having 

> 00097fec T _ZN7vsomeip12payload_implC1Ev

when i `arm-nm libvsomeip-sd.so` i'm having          

> U _ZN7vsomeip12payload_implC1Ev

maiminhtien94, 2018-05-21:
Yeb! I am too. When run on PC ubuntu it ok. But run on embedded board still show error. but the data  transaction is correct.

gaetanprin, 2018-05-21:
Maybe data transaction is ok when you’re running out of network but service discovery provided by libvsomeip-sd.so is not working