COVESA/vsomeip #11: Multiple vsomeip services configuration with secutiry

majanko, 2017-10-25:
Description of configuration I work on:
we have multiple services on linux device. Let‘s say:

- TheFirstService – the_first_service vsomeip name.
- TheSecondService – the_second_service vsomeip name

We‘re using local unix sockets with security switched on to control the access.
Client connects from the docker image. I think this is no so important. The client just needs to have access to unix socket files as an client started directly on linux device. 

It is not clear to me whether I need to have vsomeipd daemon started. Except both services.
Do I need to write separate configuration files for each service and its clients? For example:
/etc/vsomeip/vsomeip-first-service.json
/etc/vsomeip/vsomeip-second-service.json

I‘m able to connect to the first one from client using configuration inside just one config file /etc/vsomeip/vsomeip-services.json. But just to one of them. The second one refuses to offer a service durring the startup.
So, the main question is how to correctly configure multiple vsomeip services using local unix sockets with security implementation based on UNIX credentials.

- Do I need configuration file for each service?
- Do I need started vsomeipd daemon vith its own configuration(e.g. /etc/vsomeip.json)?

My configuration files are in attached [vsomeipValeo.tar.gz](https://github.com/GENIVI/vsomeip/files/1413568/vsomeipValeo.tar.gz)
 archive:
configs/vsomeip–services.json – multiple services configuration in one config
configs/vsomeip–first–service.json – the first service configuration
configs/vsomeip–second–service.json – the second service configuration
logs/TheFirstServiceLog.txt – FirstService console output.
logs/TheSecondServiceLog.txt – SecondService console output.
logs/ServiceExampleConnectionLog.txt – Client console output.

gunnarx, 2017-10-25:
I find that emails are usually more effective in sorting out these kinds of questions.  The ability to quote parts and have a back-and-forth discussion is really useful.   

Also, more frequently people will review or search mailing list archives for questions already answered, more than the list of closed GitHub issues.  Traditionally the list [genivi-ipc@lists.genivi.org](https://lists.genivi.org) was used for IPC, CommonAPI, Franca IDL and (I think) SOME-IP.  

If @GenivivSOMEIPmaintainer agrees, I'd recommend placing such questions on the mailing list.  What do you think?

majanko, 2017-10-26:
I agree I'll do it.

gunnarx, 2018-02-05:
@majanko Was this processed on the mailing list?  Can you or the maintainer report on the outcome and close this issue?

GenivivSOMEIPmaintainer, 2018-05-23:
We tested the given configuration without problems.

closed 2018-05-23 by GenivivSOMEIPmaintainer

Prasad454, 2019-07-26:
@majanko : Can you pls share the config file how you configure Docker container to use it as an application for vsomeip communication