COVESA/vsomeip #13: Compile error on x64-Linux

fefe17, 2017-11-22:
First a warning:

vsomeip/implementation/helper/boost/asio/detail/handler_type_requirements_ext.hpp:223:41: warning: conversion to ‘int’ from ‘long unsigned int’ may alter its value [-Wconversion]
vsomeip/implementation/helper/boost/asio/basic_datagram_socket_ext.hpp:347:5: note: in expansion of macro ‘BOOST_ASIO_WRITE_HANDLER_CHECK’

(several instances of that)

Then compile errors:

In file included from vsomeip/implementation/endpoints/src/netlink_connector.cpp:13:0:
vsomeip/implementation/endpoints/src/../include/netlink_connector.hpp: In function ‘bool vsomeip::operator==(const vsomeip::nl_endpoint<Protocol>&, const vsomeip::nl_endpoint<Protocol>&)’:
vsomeip/implementation/endpoints/src/../include/netlink_connector.hpp:101:28: error: no match for ‘operator==’ (operand types are ‘const sockaddr_nl’ and ‘const sockaddr_nl’)
         return e1.sockaddr == e2.sockaddr;
                ~~~~~~~~~~~~^~~~~~~~~~~~~~
vsomeip/implementation/endpoints/src/../include/netlink_connector.hpp: In function ‘bool vsomeip::operator!=(const vsomeip::nl_endpoint<Protocol>&, const vsomeip::nl_endpoint<Protocol>&)’:
vsomeip/implementation/endpoints/src/../include/netlink_connector.hpp:108:30: error: no match for ‘operator==’ (operand types are ‘const sockaddr_nl’ and ‘const sockaddr_nl’)
         return !(e1.sockaddr == e2.sockaddr);
                  ~~~~~~~~~~~~^~~~~~~~~~~~~~
vsomeip/implementation/endpoints/src/../include/netlink_connector.hpp: In function ‘bool vsomeip::operator<(const vsomeip::nl_endpoint<Protocol>&, const vsomeip::nl_endpoint<Protocol>&)’:
vsomeip/implementation/endpoints/src/../include/netlink_connector.hpp:115:28: error: no match for ‘operator<’ (operand types are ‘const sockaddr_nl’ and ‘const sockaddr_nl’)
         return e1.sockaddr < e2.sockaddr;
                ~~~~~~~~~~~~^~~~~~~~~~~~~
vsomeip/implementation/endpoints/src/../include/netlink_connector.hpp: In function ‘bool vsomeip::operator>(const vsomeip::nl_endpoint<Protocol>&, const vsomeip::nl_endpoint<Protocol>&)’:
vsomeip/implementation/endpoints/src/../include/netlink_connector.hpp:122:28: error: no match for ‘operator<’ (operand types are ‘const sockaddr_nl’ and ‘const sockaddr_nl’)
         return e2.sockaddr < e1.sockaddr;
                ~~~~~~~~~~~~^~~~~~~~~~~~~

I'm a bit puzzled why you would want to compare sockaddr_nl in the first place, to be honest.
Especially for <. That makes no sense to me. But for what it's worth, I added this code:

    static bool operator==(const struct sockaddr_nl& a,const struct sockaddr_nl& b) {
      return (a.nl_family == b.nl_family &&
              a.nl_pid == b.nl_pid &&
              a.nl_groups == b.nl_groups);
    }   

    static bool operator<(const struct sockaddr_nl& a,const struct sockaddr_nl& b) {
      if (a.nl_family < b.nl_family) return true;
      if (a.nl_family > b.nl_family) return false;
      if (a.nl_pid < b.nl_pid) return true;
      if (a.nl_pid > b.nl_pid) return false;
      return a.nl_groups < b.nl_groups;
    }   

that made it compile.

ykhokhulya, 2017-11-29:
This issue related to [FTBS using gcc 7](https://github.com/GENIVI/vsomeip/issues/4).
Seems in next release (2.7.1) it will be fixed.

gunnarx, 2018-02-05:
Thanks @ykhokhulya.  Yes I think it is covered by a recent pull request #14.
@frznlogic - could you answer @fefe17 - the < operator was used for sorting the items in a list?

GenivivSOMEIPmaintainer, 2018-02-07:
Closing, as v2.7.1 is available.

closed 2018-02-07 by GenivivSOMEIPmaintainer