COVESA/vsomeip #16: Bug in the service discovery (SD) implementation

p7mX, 2018-02-04:
Using IPv6 with UDP for service discovery (SD) is not possible because of the following bug in the SD implementation (file: service_discovery_impl.cpp - line: 1802):

https://github.com/GENIVI/vsomeip/blob/4808f3130c97cea3f0168005b9df029282060539/implementation/service_discovery/src/service_discovery_impl.cpp#L1802

Using "its_unreliable->get_local_port()" instead of "its_reliable->get_local_port()" fixes the bug.

GenivivSOMEIPmaintainer, 2018-02-07:
Taken over for 2.10.12. Thanks.

gunnarx, 2018-02-07:
Oh, that's a nasty "copy-paste" bug there - good catch.  

Yeah, repetition leads to bugs... I wonder if some more refactoring might not be possible here?
I see it can be hard to generalize all the template instantiations using a temp variable would also have helped here:    port = its_unreliable->get_local_port(), and then in the two find statements, reuse "port".)   That's one small repetition, but would also remove the bug.
Like it's done already on line 873 and 895, by the way: https://github.com/GENIVI/vsomeip/blob/4808f3130c97cea3f0168005b9df029282060539/implementation/service_discovery/src/service_discovery_impl.cpp#L873

Then there's repetition of the structures under the two if's.  It seems like the 2 x 2 calls to find_existing_option() could be reduced to 2 only, by extract method refactoring, if we wanted to.

Also, I wonder if a static analysis tool would not have have caught this because it seems that its_reliable could be NULL (or whatever is the iterator equivalent here), the way the code was written now.

@GenivivSOMEIPmaintainer - when will 2.10.12 be released?  I would recommend that such fixes are merged immediately to master.  (Or if you prefer, there could be a separate branch, like **develop** for such fixes - as long as it is documented so that contributors and users know)

(@p7mX - you could have sent a pull request to make that quicker, but as somebody interested in this component - thanks for reporting anyway.)

selbstereg, 2018-04-18:
Can someone give me a hint, how to configure vsomeip to use IPv6 for service discovery?

GenivivSOMEIPmaintainer, 2018-05-23:
We do not yet have any experience using vsomeip in IPv6 networks. In theory, you only need to provide IPv6 addresses for the unicast as well as the multicast address within the service_discovery node.

closed 2018-05-23 by GenivivSOMEIPmaintainer