COVESA/vsomeip #17: Building vsomeip for 32 bit architechture

vkram72, 2018-02-06:
Can you provide the procedure for building it for 32 bit architecture in Ubuntu 16. I tried adding CXX flags as -m32, but I am facing issues with boost libraries

GenivivSOMEIPmaintainer, 2018-02-07:
Do you build on Ubuntu 32-bit or 64-bit?

vkram72, 2018-02-08:
Ubuntu 64-bit

GenivivSOMEIPmaintainer, 2018-02-09:
# this two steps weren't neccessary on my machine, so first try without them
#sudo dpkg --add-architecture i386
#sudo apt-get update
sudo apt-get install libboost-system1.58-dev:i386 libboost-thread1.58-dev:i386 libboost-log1.58-dev:i386 libsystemd-dev:i386
mkdir build32bit
cd build32bit
cmake -DCMAKE_BUILD_TYPE=Release \
-DCMAKE_CXX_FLAGS_RELEASE='-O2 -g -Wall -Wextra -Wpedantic -Wconversion -m32' \
-DBOOST_LIBRARYDIR=/usr/lib/i386-linux-gnu \
..