COVESA/vsomeip #18: Error building with latest Boost 1.66

samup4web, 2018-02-13:
The latest boost library comes with interface changes that are not backward compatible with VsomeIp implementation. The VSomeIp README file says supported boost version is => 1.55,  but this is not valid at the moment.  

Below are some errors from my build attempt:

`[  1%] Building CXX object CMakeFiles/vsomeip.dir/implementation/endpoints/src/endpoint_impl.cpp.o
In file included from /home/samueli/workspace/vsomeip/implementation/helper/boost/asio/datagram_socket_service_ext.hpp:33:0,
                 from /home/samueli/workspace/vsomeip/implementation/helper/boost/asio/basic_datagram_socket_ext.hpp:24,
                 from /home/samueli/workspace/vsomeip/implementation/helper/boost/asio/ip/udp_ext.hpp:20,
                 from /home/samueli/workspace/vsomeip/implementation/endpoints/src/endpoint_impl.cpp:8:
/home/samueli/workspace/vsomeip/implementation/helper/boost/asio/detail/reactive_socket_service_ext.hpp:29:10: fatal error: boost/asio/detail/addressof.hpp: No such file or directory
 #include <boost/asio/detail/addressof.hpp>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~`

`make[2]: *** [CMakeFiles/vsomeip.dir/build.make:135: CMakeFiles/vsomeip.dir/implementation/endpoints/src/endpoint_impl.cpp.o] Error 1
[  2%] Building CXX object CMakeFiles/vsomeip.dir/implementation/endpoints/src/local_client_endpoint_impl.cpp.o
/home/samueli/workspace/vsomeip/implementation/endpoints/src/local_client_endpoint_impl.cpp: In member function ‘virtual void vsomeip::local_client_endpoint_impl::connect()’:
/home/samueli/workspace/vsomeip/implementation/endpoints/src/local_client_endpoint_impl.cpp:128:64: error: ‘class boost::asio::basic_stream_socket<boost::asio::local::stream_protocol>’ has no member named ‘native’
                         credentials::send_credentials(socket_->native(),
`

`[  4%] Building CXX object CMakeFiles/vsomeip.dir/implementation/endpoints/src/local_server_endpoint_impl.cpp.o
/home/samueli/workspace/vsomeip/implementation/endpoints/src/local_server_endpoint_impl.cpp: In constructor ‘vsomeip::local_server_endpoint_impl::local_server_endpoint_impl(std::shared_ptr<vsomeip::endpoint_host>, vsomeip::server_endpoint_impl<boost::asio::local::stream_protocol>::endpoint_type, boost::asio::io_service&, uint32_t, uint32_t, vsomeip::configuration::endpoint_queue_limit_t)’:
/home/samueli/workspace/vsomeip/implementation/endpoints/src/local_server_endpoint_impl.cpp:52:53: error: ‘boost::asio::local::stream_protocol::acceptor {aka class boost::asio::basic_socket_acceptor<boost::asio::local::stream_protocol>}’ has no member named ‘native’
         credentials::activate_credentials(acceptor_.native());`

`/home/samueli/workspace/vsomeip/implementation/endpoints/src/local_server_endpoint_impl.cpp:211:44: error: ‘vsomeip::server_endpoint_impl<boost::asio::local::stream_protocol>::socket_type {aka class boost::asio::basic_stream_socket<boost::asio::local::stream_protocol>}’ has no member named ‘native’
                      new_connection_socket.native(), uid, gid);
                                            ^~~~~~
`

mighty84, 2018-04-05:
Hi,
I patched it up real quick to make it work:
https://bitbucket.org/Mighty84/vsomeip_boost_1.66
I also built a patch for vsomeip-2.6.2 if that's any help to you.

I wasn't able to get it to work with the custom boost::asio::ip::udp_ext
helper so I switched it to Boosts own boost::asio::ip::udp. Not really sure 
what the compiler is complaining about.

samup4web, 2018-04-18:
@mighty84 Thanks for the help. Although we settled with Boosts 1.64, I may revisit this soon.

GenivivSOMEIPmaintainer, 2018-05-23:
We will not support boost >= 1.66 before vsomeip 3.0

closed 2018-05-23 by GenivivSOMEIPmaintainer

samup4web, 2018-07-31:
I guess the information on dependency under the build instruction should be updated accordingly.

zeerd, 2020-03-05:
@mighty84 Hello! Buddy, the link you given was invalid now, any other way to get it?

huangbilly, 2021-02-13:
I think I found that link: https://bitbucket.org/Mighty84/vsomeip/commits/cdb1125c0ab1c4230159c92badef38957436409d