COVESA/vsomeip #23: Questions about TLV support

nhkim1104, 2018-05-10:
Hello,

Does vsomeip support a TLV(Type-Length-Value) encoding scheme? 
If not, is there a support plan?

GenivivSOMEIPmaintainer, 2018-05-23:
Please explain in greater detail what this is about.

nhkim1104, 2018-05-31:
Hello, 

TLV (type-length-value or tag-length-value) is a kind of data encoding scheme.
For simple example;
- type = AE 01 0A
- length = 04
- value = 12 34 00 56  
If we want to send  data we can make it like (AE 01 0A 04 12 34 00 56).

For more detailed information please refer to wiki page below. I think it is better to see wiki than I explain it.
=> https://en.wikipedia.org/wiki/Type-length-value

Thanks

GenivivSOMEIPmaintainer, 2018-06-05:
The SOME/IP protocol does not support TLV. If you need extensible data structures, you can use unions and/or extensible structs. Please note that vsomeip does not handle the message payload, this is done by [CommonAPI-SomeIP](https://github.com/GENIVI/capicxx-someip-runtime).

closed 2018-06-05 by GenivivSOMEIPmaintainer

nhkim1104, 2018-06-08:
Thanks for your response. As you commented, I will refer the CommonAPI-SomeIP.