COVESA/vsomeip #25: Failed build with gcc8 compiler

ykhokhulya, 2018-05-22:
```
/home/ykhokhulya/work/packages/vSomeIP/implementation/endpoints/src/../include/netlink_connector.hpp: In member function ‘vsomeip::nl_endpoint<Protocol>::data_type* vsomeip::nl_endpoint<Protocol>::data()’:
/home/ykhokhulya/work/packages/vSomeIP/implementation/endpoints/src/../include/netlink_connector.hpp:70:17: error: cannot convert ‘sockaddr_nl*’ to ‘vsomeip::nl_endpoint<Protocol>::data_type*’ {aka ‘sockaddr*’} in return
         return &sockaddr;
                 ^~~~~~~~
```
GCC version: 8.1.0

ykhokhulya, 2018-05-22:
The following patch fixes the issue:
```
diff --git a/implementation/endpoints/include/netlink_connector.hpp b/implementation/endpoints/include/netlink_connector.hpp
index f71ba88..730874e 100644
--- a/implementation/endpoints/include/netlink_connector.hpp
+++ b/implementation/endpoints/include/netlink_connector.hpp
@@ -67,7 +67,7 @@ public:
     /// Get the underlying endpoint in the native type.
     data_type* data()
     {
-        return &sockaddr;
+        return reinterpret_cast<struct sockaddr*>(&sockaddr);
     }
 
     /// Get the underlying endpoint in the native type.
```

GenivivSOMEIPmaintainer, 2018-05-23:
Applied. Will be part of v2.10.22.

closed 2018-05-23 by GenivivSOMEIPmaintainer