COVESA/vsomeip #26: No offer service multicast on Windows

azeps, 2018-05-29:
Hello.

Under Linux vsomeip works as expected but I have difficulties as we also would like to use this library together with CommonAPI on Windows.
The problem I'm experiencing is the lack of OFFER_SERVICE multicast messages. There are simply not sent while in contrast to that FIND_SERVICE messages are visible using wireshark.

I've tested using Win 7 (64 Bit) and Win 10.
All was compiled using Visual Studio 14 (2015).
I have used Boost 1.61 for this.

Is this a common problem? As it was buildable for Windows I've assumed that this port is also runnable.
I even checked multicast settings but as the FIND_SERVICE messages are correctly sent using multicast I assume this is not the problem.

My project is based on the examples I could find here:
capicxx-someip-tools/tree/master/CommonAPI-Examples

azeps, 2018-05-29:
I would like to add something to make it easier to reproduce.
It's not needed to actually use the full functionality with CommonAPI to discover the problem.
Even the examples coming with vsomeip have the issue.

Let's say I execute this:
set VSOMEIP_CONFIGURATION=vsomeip-udp-client.json
set VSOMEIP_APPLICATION_NAME=client-sample
subscribe-sample.exe

4 FIND_SERVICE frames are clearly visible on my ethernet.

But if I execute this:
set VSOMEIP_CONFIGURATION=vsomeip-udp-service.json
set VSOMEIP_APPLICATION_NAME=service-sample
notify-sample.exe

I don't get the expected frequent OFFER_SERVICE frames that I can see with Linux.

007herelsp, 2018-08-13:
I found the cause of the problem and fixed it

007herelsp, 2018-08-14:
The following code is available in the routing_manager_impl.cpp
void routing_manager_impl::start() {
#ifndef _WIN32
    netlink_connector_ = std::make_shared<netlink_connector>(host_->get_io(),
            configuration_->get_unicast_address(),
            boost::asio::ip::address::from_string(configuration_->get_sd_multicast()));
    netlink_connector_->register_net_if_changes_handler(
            std::bind(&routing_manager_impl::on_net_interface_or_route_state_changed,
            this, std::placeholders::_1, std::placeholders::_2, std::placeholders::_3));
    netlink_connector_->start();
#else
    {
        std::lock_guard<std::mutex> its_lock(pending_sd_offers_mutex_);
        start_ip_routing();
    }
#endif

Modify the void routing_manager_impl::start_ip_routing() as follows:

void routing_manager_impl::start_ip_routing() {

    if_state_running_ = true; //Add
	
    if (discovery_) {
        discovery_->start();
    } else {
        init_routing_info();
    }

    for (auto its_service : pending_sd_offers_) {
        init_service_info(its_service.first, its_service.second, true);
    }
    pending_sd_offers_.clear();

    routing_running_ = true;
    VSOMEIP_INFO << VSOMEIP_ROUTING_READY_MESSAGE;
}

8vp, 2018-09-28:
Hi, 

Is this an open issue? I am new here, and I am trying to create a service and client following example in the wiki page. All looks fine to start with and I can see that the client subscribes to the service as expected. Not seeing any OFFER message from the service side, why is that? Please advise.

aananthcn, 2019-01-16:
> The following code is available in the routing_manager_impl.cpp
> void routing_manager_impl::start() {
> #ifndef _WIN32 netlink_connector_ = std::make_shared<netlink_connector>(host_->get_io(), [...]

Hi,

I spent almost a week to find that the attributes of serviceinfo class (particularly reliable_ and unreliable_ endpoints) are 0. And due to this the offer services are not going out of the non-Linux node. But when I applied the changes from 007herelsp, it worked!! I am not sure if the change provided by 007herelsp is a fix or work-around, but please update the code.

Also I couldn't locate "external_local_routing_test_service.json" file under test folder. When I looked into test/routing_tests/conf/external_local_routing_test_service.json.in file, I could see that service-discovery is disabled. I don't know if the external routing test passed (for both Linux and non-Linux) nodes.

gunnarx, 2019-01-31:
@007herelsp  I would propose for a code change that you post a diff instead of only the final result.  
A git diff will show also the original code so that it's not necessary to search for it, and also what version of the code you started with, since this can change over time.  

Eventually after discussion, if you truly want propose a code change, it should be a pull request of course.

gunnarx, 2019-01-31:
@8vp You are asking for general help to "get going".  The standard answer is to look at documentation and try the examples directory first.  After that, a mailing list question is ideal.  But unfortunately the maintainer has not provided mailing-list information in the README so at the moment I can't refer you to it.

In any case I'd recommend you ask your Linux-related question separately, and not in this ticket that is about a specific Windows issue.  It's almost guaranteed that you have a configuration problem and not anything related to this issue.

gunnarx, 2019-01-31:
@GenivivSOMEIPmaintainer Please evaluate and answer the ticket.

lutzbichler, 2019-07-24:
The fix for Windows will be included in vsomeip v3.

closed 2020-07-31 by lutzbichler