COVESA/vsomeip #27: No usage examples and configuration files for e2e protected

007herelsp, 2018-06-06:
I tried to use the E2E protection feature, but I didn't find the relevant usage examples and configuration files, which caused me to not use the feature

JaehoonLeee, 2018-07-06:
I also tried to use the e2e protection. But, e2e protection feature is difficult to use.
Please , share example configuration to us ?

gunnarx, 2019-01-31:
@GenivivSOMEIPmaintainer Please answer the ticket.

lutzbichler, 2020-08-11:
Please take a look into test/e2e_tests/conf. It contains examples of howto configure messages to be protected/checked.

closed 2020-08-11 by lutzbichler

GHARBIAchref, 2024-09-21:
Hello @fengmao31  @duartenfonseca  
 I want to use end-to-end protection in my application. Is using end-to-end protection done solely through configuration in .json configuration files on both the client and server sides of the services, or does it also require code implementation within my algorithm? And how can I verify if the E2E protection profile 4 is working correctly in Wireshark? 
 thanks

fengmao31, 2024-09-21:
> Hello @lutzbichler I want to use end-to-end protection in my application. Is using end-to-end protection done solely through configuration in .json configuration files on both the client and server sides of the services, or does it also require code implementation within my algorithm? And how can I verify if the E2E protection profile 4 is working correctly in Wireshark? thanks

You donnot need code with your algorithm. Mostly adas standards vsomeip finish for you, just config it.

fengmao31, 2024-09-21:
You can see the byte in the network package. And you can caculate the e2e part in the header. Of course, if you have autosar tool, it will be more easy for you. The tool can show the structure of the message.