COVESA/vsomeip #29: Not able to set up communication between two devices

uj864, 2018-07-23:
I followed the instructions given in VSOMEIP in 10 and its is working fine on one host. But it is not working when I am trying to do it on two ubuntu machines.
My files are:
World.cpp
#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678

std::shared_ptr< vsomeip::application > app;

int main() {

    app = vsomeip::runtime::get()->create_application("World");
    app->init();
    app->offer_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID);
    app->start();
}

Hello.cpp
#include <iomanip>
#include <iostream>

#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678

std::shared_ptr< vsomeip::application > app;

void on_availability(vsomeip::service_t _service, vsomeip::instance_t _instance, bool _is_available) {
    std::cout << "Service ["
            << std::setw(4) << std::setfill('0') << std::hex << _service << "." << _instance
            << "] is " << (_is_available ? "available." : "NOT available.")  << std::endl;
}

int main() {

    app = vsomeip::runtime::get()->create_application("Hello");
    app->init();
    app->register_availability_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, on_availability);
    app->request_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID);
    app->start();
}

I used the same json files as in the tutorial(with unicast changed according to my machines) and named them s.json and c.json.

I have the executables and the json files in the same folder from that folder I used the following commands:

env VSOMEIP_CONFIGURATION=./s.json VSOMEIP_APPLICATION_NAME=World ./World (on host 1)
env VSOMEIP_CONFIGURATION=./c.json VSOMEIP_APPLICATION_NAME=Hello ./Hello (on host 2)

I also used the command: route add -nv 224.224.224.245 dev eth0

But the applications keep running and do not communicate with each other.

Outpost on host1:

2018-07-23 17:36:41.523536 [info] Parsed vsomeip configuration in 2ms
2018-07-23 17:36:41.523640 [info] Using configuration file: "s.json".
2018-07-23 17:36:41.523669 [info] Default configuration module loaded.
2018-07-23 17:36:41.523680 [info] Initializing vsomeip application "World".
2018-07-23 17:36:41.523737 [warning] Routing Manager seems to be inactive. Taking over...
2018-07-23 17:36:41.523832 [info] SOME/IP client identifier configured. Using 0009 (was: 1212)
2018-07-23 17:36:41.523840 [info] Instantiating routing manager [Host].
2018-07-23 17:36:41.523887 [info] init_routing_endpoint Routing endpoint at /tmp/vsomeip-0
2018-07-23 17:36:41.523936 [info] Client [9] is connecting to [0] at /tmp/vsomeip-0
2018-07-23 17:36:41.523946 [info] Service Discovery enabled. Trying to load module.
2018-07-23 17:36:41.527714 [info] Service Discovery module loaded.
2018-07-23 17:36:41.527828 [info] Application(World, 9) is initialized (11, 100).
2018-07-23 17:36:41.527843 [info] OFFER(0009): [1234.5678:0.0]
2018-07-23 17:36:41.528033 [info] Starting vsomeip application "World" using 2 threads
2018-07-23 17:36:41.528956 [info] main dispatch thread id from application: 0009 (World) is: 7ffff59b2700 TID: 13161
2018-07-23 17:36:41.528982 [info] shutdown thread id from application: 0009 (World) is: 7ffff3978700 TID: 13162
2018-07-23 17:36:41.529005 [info] Watchdog is disabled!
2018-07-23 17:36:41.529070 [info] io thread id from application: 0009 (World) is: 7ffff7fcd740 TID: 13160
2018-07-23 17:36:41.529126 [info] io thread id from application: 0009 (World) is: 7fffdffff700 TID: 13164
2018-07-23 17:36:41.529206 [info] vSomeIP 2.10.21 | (default)
2018-07-23 17:36:41.529281 [info] Network interface "eth0" state changed: up
2018-07-23 17:36:41.529325 [info] Route "default route (0.0.0.0/0) if: eth0 gw: 10.24.34.1" state changed: up
2018-07-23 17:36:41.529421 [info] SOME/IP routing ready.
2018-07-23 17:36:41.529488 [warning] Route "224.224.224.245/32 if: eth0 gw: n/a" state changed: up
2018-07-23 17:36:44.729077 [warning] Releasing client identifier 1212. Its corresponding application went offline while no routing manager was running.
2018-07-23 17:36:44.729153 [info] Application/Client 1212 is deregistering.
Outpost on host2:
\2018-07-23 17:37:31.607453 [info] Parsed vsomeip configuration in 1ms
2018-07-23 17:37:31.607502 [info] Using configuration file: "c.json".
2018-07-23 17:37:31.607511 [info] Default configuration module loaded.
2018-07-23 17:37:31.607520 [info] Initializing vsomeip application "Hello".
2018-07-23 17:37:31.607547 [warning] Routing Manager seems to be inactive. Taking over...
2018-07-23 17:37:31.607570 [info] SOME/IP client identifier configured. Using 000a (was: 1313)
2018-07-23 17:37:31.607576 [info] Instantiating routing manager [Host].
2018-07-23 17:37:31.607610 [info] init_routing_endpoint Routing endpoint at /tmp/vsomeip-0
2018-07-23 17:37:31.607637 [info] Client [a] is connecting to [0] at /tmp/vsomeip-0
2018-07-23 17:37:31.607645 [info] Service Discovery enabled. Trying to load module.
2018-07-23 17:37:31.608834 [info] Service Discovery module loaded.
2018-07-23 17:37:31.608866 [info] Application(Hello, a) is initialized (11, 100).
2018-07-23 17:37:31.608882 [info] REQUEST(000a): [1234.5678:255.4294967295]
2018-07-23 17:37:31.608941 [info] Starting vsomeip application "Hello" using 2 threads
2018-07-23 17:37:31.609086 [info] Watchdog is disabled!
2018-07-23 17:37:31.609264 [info] shutdown thread id from application: 000a (Hello) is: 7ffff3978700 TID: 13584
2018-07-23 17:37:31.609291 [info] io thread id from application: 000a (Hello) is: 7ffff7fcd740 TID: 13582
2018-07-23 17:37:31.609371 [info] io thread id from application: 000a (Hello) is: 7fffe7fff700 TID: 13586
2018-07-23 17:37:31.609381 [info] main dispatch thread id from application: 000a (Hello) is: 7ffff59b2700 TID: 13583
2018-07-23 17:37:31.609390 [info] vSomeIP 2.10.21 | (default)
Service [1234.5678] is NOT available.
2018-07-23 17:37:31.609509 [info] Network interface "eth0" state changed: up
2018-07-23 17:37:31.609539 [info] Route "default route (0.0.0.0/0) if: eth0 gw: 10.24.34.1" state changed: up
2018-07-23 17:37:31.609645 [info] SOME/IP routing ready.
2018-07-23 17:37:31.609664 [warning] Route "224.224.224.245/32 if: eth0 gw: n/a" state changed: up
2018-07-23 17:37:34.809112 [warning] Releasing client identifier 0009. Its corresponding application went offline while no routing manager was running.
2018-07-23 17:37:34.809141 [info] Application/Client 0009 is deregistering.

hemalbavishi, 2018-07-26:
Hello,

Follow the instruction provided in the readme.txt to run application on two different devices. Please ensure that you configure host IP address in the corresponding JSON file before executing the same. 

https://github.com/GENIVI/vsomeip/tree/master/examples 

Cheers!

brenkem, 2018-10-15:
The solution of this problem can also be found here: https://github.com/GENIVI/vsomeip/issues/36

gunnarx, 2019-01-31:
@sachindes I'm not equipped to identify the problem exactly for you, but some questions that helped me in the past:
- Do you need to / forgot to set up [multicast routing](https://github.com/GENIVI/vsomeip/blob/master/documentation/multicast.txt) on one or both targets?  
- Are you running a separate process (e.g. vsomeipd) as routing manager or relying on the applications to set up their own?

gunnarx, 2019-01-31:
Reopened ticket for the benefit of @sachindes.

I would recommend you set up logging at debug level and post relevant parts of the log as "evidence" of what you wrote in text "...client not acknowledge Service is Available and initiate request." 
Probably easier for an expert (not me) to make a judgement by looking at the exact sequence.

Obviously trying to reduce your problem to run the simple examples first will also help in getting support (or you will find the problem yourself).

NandhanThiravia, 2019-02-11:
Hi @gunnarx 

I am associated with @sachindes  and I would like to follow-up on the issue. So as per @sachindes, the communication is happening between a Ubuntu PC and an Android Target, but the same code and same setup is not working with 2 Android Targets, i.e. both the targets are not able to discover each other.

Here are the Client & Server Code:
[src.zip](https://github.com/GENIVI/vsomeip/files/2850961/src.zip)

Here are the JSON files used:
[config.zip](https://github.com/GENIVI/vsomeip/files/2850968/config.zip)

And here are the logs as requested by you:
[vsomeip_client.log](https://github.com/GENIVI/vsomeip/files/2850948/vsomeip_client.log)
[vsomeip_service.log](https://github.com/GENIVI/vsomeip/files/2850949/vsomeip_service.log)

Any help is appreciable.

Regards
Nandhan

NandhanThiravia, 2019-02-11:
I have tried the below link too but it is not working for me:
[https://github.com/GENIVI/vsomeip/issues/36](url)

brenkem, 2019-02-11:
Sounds tricky.

I also only used it to connect a PC to a target.

But did you double check:
1. the routes and the traffic (via wireshark)
2. pass the "--tcp" option from #32 
3. I would suggest to test one of the example .json files under config/
vsomeip-tcp-client.json
vsomeip-tcp-service.json

Best regards
Maik

sachindes, 2019-02-11:
I do work with Nandhan Thiravia. We had tried --tcp and can see multi cast
route being set in Ip table and log  Here PC setup is a virtual machine
with windows 10 as guest OS
Same example worked with another setup( VM Ubuntu and target) already. We
will try suggestion given by you again.
However in general we notice reliability issues. Specially we faced issues
between inter target someip communication.

Can anyone also suggest how can we have more reliable behaviour. Traffic is
noticed at both end( Service discovery) but don't know why request is not
initiated by client.

NandhanThiravia, 2019-02-12:
Hey All

Got vsomeip over capi working, the challenges faced are mentioned below:
1) The Client Android target that I was using was not receiving the Multicast Packets that the Service Android target was sending.
2) Once I connected the Client Android Target through LAN, the multicast packets started receiving and Client was able to communicate with Service.

I read somewhere in the blogs that some OEMs disable the send/receive of Multicast packets for battery optimization.

Also, I used the basic Client.json & Service.json with no complex features in them. I even did not use "--tcp" and it still worked without any more hazzle.

Thank you all for helping :)

gunnarx, 2019-02-12:
Interesting, and thanks for sharing the result!  

> 2\. Once I connected the Client Android Target through LAN

How do you mean it was connected before that?

closed 2019-07-22 by lutzbichler

Siddharthav1, 2020-01-23:
@NandhanThiravia  sir,

  I am using CPI SOME/IP from https://genivi-oss.atlassian.net/wiki/spaces/COMMONAPICPP/pages/5472320/CommonAPI+C+SOME+IP+in+10+minutes. Everything is good on a single machine but twi PC communication is happening. May I know what wrong am doing here. please.

[someip.zip](https://github.com/GENIVI/vsomeip/files/4101160/someip.zip)

I have attached the files that I am using right now. Any suggestion will help help me a lot.

pjacky03, 2022-07-14:
> Everything is good on a single machine but twi PC communication is happening. May I know what wrong am doing here. please. (vsomeip two device communication is working without cp

I faced the same problem. I used vsomeip helloworld sample and two device communication is working without cpi.
But when I used the following commonapi sample, the connection can not be established. 
https://github.com/COVESA/capicxx-core-tools/tree/master/CommonAPI-Examples/E01HelloWorl

I attached the configuration filed I used. Does anyone know how to fix it ? Thank you for help.
[E01HelloWorld.zip](https://github.com/COVESA/vsomeip/files/9108908/E01HelloWorld.zip)

pjacky03, 2022-07-19:
> > Everything is good on a single machine but twi PC communication is happening. May I know what wrong am doing here. please. (vsomeip two device communication is working without cp
> 
> I faced the same problem. I used vsomeip helloworld sample and two device communication was working without cpi. But when I used the following commonapi sample, the connection can not be established. https://github.com/COVESA/capicxx-core-tools/tree/master/CommonAPI-Examples/E01HelloWorld
> 
> I attached the configuration filed I used. Does anyone know how to fix it ? Thank you for help. [E01HelloWorld.zip](https://github.com/COVESA/vsomeip/files/9108908/E01HelloWorld.zip)

I found the solution. Thank you

aadarshkt, 2024-10-29:
> on't know why client not acknowledge Servic

Same issue, at client side, I can see both find and offer packets but next three-way handshake is not being started within my two EC2 instances.