COVESA/vsomeip #31: vsomeip does not compile with boost 1.66

frznlogic, 2018-07-31:
Hi,

I'm porting meta-ivi to yocto sumo release and they have bumped versions of boost to version 1.66 and I've run into some issues with compiling vsomeip with that version of boost. From what I can gather they changed some of the header files of asio among other things. Here's an example:

```

| FAILED: CMakeFiles/vsomeip-sd.dir/implementation/service_discovery/src/service_discovery_impl.cpp.o
| /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++  -DVSOMEIP_VERSION=\"2.10.10\" -Dvsomeip_sd_EXPORTS -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/interface -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot/usr/include -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/include -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/helper -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot/usr/include/dlt -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0=/usr/src/debug/vsomeip/2.10.21-r0 -fdebug-prefix-map=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot= -fdebug-prefix-map=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot -DUSE_DLT -DLINUX -D_GLIBCXX_USE_NANOSLEEP -pthread -O -Wall -Wextra -Wformat -Wformat-security -Wconversion -fexceptions -fstrict-aliasing -fstack-protector -fasynchronous-unwind-tables -fno-omit-frame-pointer -D_FORTIFY_SOURCE=2 -DBOOST_LOG_DYN_LINK -g  -std=c++11  -Wl,-export-dynamic -Wl,--version-script=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/exportmap.gcc -O2 -g -DNDEBUG -fPIC -MD -MT CMakeFiles/vsomeip-sd.dir/implementation/service_discovery/src/service_discovery_impl.cpp.o -MF CMakeFiles/vsomeip-sd.dir/implementation/service_discovery/src/service_discovery_impl.cpp.o.d -o CMakeFiles/vsomeip-sd.dir/implementation/service_discovery/src/service_discovery_impl.cpp.o -c /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/service_discovery/src/service_discovery_impl.cpp
| In file included from /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/helper/boost/asio/datagram_socket_service_ext.hpp:33:0,
|                  from /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/helper/boost/asio/basic_datagram_socket_ext.hpp:24,
|                  from /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/helper/boost/asio/ip/udp_ext.hpp:20,
|                  from /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/service_discovery/src/../../endpoints/include/udp_server_endpoint_impl.hpp:10,
|                  from /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/service_discovery/src/service_discovery_impl.cpp:31:
| /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/helper/boost/asio/detail/reactive_socket_service_ext.hpp:29:10: fatal error: boost/asio/detail/addressof.hpp: No such file or directory
|  #include <boost/asio/detail/addressof.hpp>
|           ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
| compilation terminated.
```

```
| FAILED: CMakeFiles/vsomeip.dir/implementation/endpoints/src/netlink_connector.cpp.o
| /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++  -DVSOMEIP_VERSION=\"2.10.10\" -Dvsomeip_EXPORTS -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/interface -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot/usr/include -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/include -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/helper -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot/usr/include/dlt -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0=/usr/src/debug/vsomeip/2.10.21-r0 -fdebug-prefix-map=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot= -fdebug-prefix-map=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot -DUSE_DLT -DLINUX -D_GLIBCXX_USE_NANOSLEEP -pthread -O -Wall -Wextra -Wformat -Wformat-security -Wconversion -fexceptions -fstrict-aliasing -fstack-protector -fasynchronous-unwind-tables -fno-omit-frame-pointer -D_FORTIFY_SOURCE=2 -DBOOST_LOG_DYN_LINK -g  -std=c++11  -Wl,-export-dynamic -Wl,--version-script=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/exportmap.gcc -O2 -g -DNDEBUG -fPIC -MD -MT CMakeFiles/vsomeip.dir/implementation/endpoints/src/netlink_connector.cpp.o -MF CMakeFiles/vsomeip.dir/implementation/endpoints/src/netlink_connector.cpp.o.d -o CMakeFiles/vsomeip.dir/implementation/endpoints/src/netlink_connector.cpp.o -c /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/endpoints/src/netlink_connector.cpp
| In file included from /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/endpoints/src/netlink_connector.cpp:14:0:
| /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/endpoints/src/../include/netlink_connector.hpp:136:46: error: expected ')' before '&' token
|      netlink_connector(boost::asio::io_service& _io, boost::asio::ip::address _address,
```

```
| FAILED: CMakeFiles/vsomeip.dir/implementation/endpoints/src/local_client_endpoint_impl.cpp.o
| /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++  -DVSOMEIP_VERSION=\"2.10.10\" -Dvsomeip_EXPORTS -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/interface -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot/usr/include -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/include -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/helper -I/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot/usr/include/dlt -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0=/usr/s [... cut ...]
| /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/endpoints/src/local_client_endpoint_impl.cpp: In member function 'virtual void vsomeip::local_client_endpoint_impl::connect()':
| /home/osan/Projects/bit/qemuarm64/build/tmp/work/aarch64-poky-linux/vsomeip/2.10.21-r0/git/implementation/endpoints/src/local_client_endpoint_impl.cpp:128:64: error: 'class boost::asio::basic_stream_socket<boost::asio::local::stream_protocol>' has no member named 'native'
|                          credentials::send_credentials(socket_->native(),

```

Currently vsomeip is unbuildable with sumo releases.

samup4web, 2018-07-31:
I had reported this problem in #18 but the authors do not plan to support Boost => 1.66

frznlogic, 2018-07-31:
@samup4web thanks for the notification, I didn't see your report, it looks almost identical to mine :D 

Basically this is blocking meta-ivi from properly bumping to sumo release, which IMHO is really bad. For now, I've temporarily deleted vsomeip from the meta-ivi images to get the build to work again.  As a note, this work is still in progress so it's not on the master branches etc yet.

GenivivSOMEIPmaintainer, 2018-07-31:
You could use the following patch as long as vsomeip 3.0 is not released.

[0001-Support-boost-v1.66.patch.zip](https://github.com/GENIVI/vsomeip/files/2244890/0001-Support-boost-v1.66.patch.zip)

frznlogic, 2018-07-31:
Thank you @GenivivSOMEIPmaintainer, that fixes my problems. I've deleted the temporary deletion and it's back in the image again with the patch ;). I'll do some further testing. I think you can close this issue for now though if you want to.

closed 2018-08-02 by GenivivSOMEIPmaintainer

OznOg, 2019-06-28:
> You could use the following patch as long as vsomeip 3.0 is not released.
> 
> [0001-Support-boost-v1.66.patch.zip](https://github.com/GENIVI/vsomeip/files/2244890/0001-Support-boost-v1.66.patch.zip)

on which revision is this patch supposed to be applied? I'm always getting conflicts

siddharthav, 2021-02-25:
Hello @GenivivSOMEIPmaintainer 

 I am trying to build the vsomeip.2.14.16. I tried to apply the patch you have attached above. but its not working. I am getting below error. Could you please help me out how can I buils 2.14.16 version please.

Below is the error that I am getting, when I try to apply patch 0001-Support-boost-v1.66.patch:

 ```
   0001-Support-boost-v1.66.patch:586: space before tab in indent.
    		io_context, protocol, native_socket)
0001-Support-boost-v1.66.patch:1314: trailing whitespace.
  
0001-Support-boost-v1.66.patch:1341: trailing whitespace.
  
0001-Support-boost-v1.66.patch:3929: trailing whitespace.
    
0001-Support-boost-v1.66.patch:3945: trailing whitespace.
        
error: implementation/configuration/include/internal.hpp: already exists in working directory
error: patch failed: implementation/endpoints/include/netlink_connector.hpp:67
error: implementation/endpoints/include/netlink_connector.hpp: patch does not apply
error: patch failed: implementation/endpoints/src/local_client_endpoint_impl.cpp:125
error: implementation/endpoints/src/local_client_endpoint_impl.cpp: patch does not apply
error: patch failed: implementation/endpoints/src/local_server_endpoint_impl.cpp:49
error: implementation/endpoints/src/local_server_endpoint_impl.cpp: patch does not apply
```

Regards,
Siddhartha V

Nitind7, 2024-10-11:
> Hello @GenivivSOMEIPmaintainer
>
> I am trying to build the vsomeip.2.14.16. I tried to apply the patch you have attached above. but its not working. I am getting below error. Could you please help me [...]

git apply --reject --whitespace=fix 0001-Support-boost-v1.66.patch [Try it]