COVESA/vsomeip #32: Unable to communicate between two HOSTs

brenkem, 2018-08-06:
I build the libraries and the examples. I was able to set up a communication on the same host. But not between two separate devices.
I configured the network as described (see [1] and [2]). But when I run [3] and [4] there is no communication over the network. also there SD 244.244.244.245 is quiet. 

[1] Service:
$ ip addr show dev enx0080c83cef81
6: enx0080c83cef81: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether 00:80:c8:3c:ef:81 brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.2/16 brd 192.168.255.255 scope global enx0080c83cef81
       valid_lft forever preferred_lft forever
    inet6 fe80::70a9:4e3a:2659:1bda/64 scope link 
       valid_lft forever preferred_lft forever

$ route
Kernel-IP-Routentabelle
Ziel            Router          Genmask         Flags Metric Ref    Use Iface
…
192.168.0.0     *               255.255.0.0     U     100    0        0 enx0080c83cef81
224.224.224.245 *               255.255.255.255 UH    0      0        0 enx0080c83cef81

$ head ../../config/vsomeip-tcp-service.json 
{
    "unicast" : "192.168.1.2",
    "netmask" : "255.255.0.0",
    "logging" :
    { 
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
        "dlt" : "false"
    },

[2] Client:
$ ip addr show dev enp0s31f6
2: enp0s31f6: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc pfifo_fast state UP group default qlen 1000
    link/ether c8:d3:ff:6c:de:9a brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.1/16 brd 192.168.255.255 scope global enp0s31f6
       valid_lft forever preferred_lft forever
    inet6 fe80::982:347a:a527:95d3/64 scope link 
       valid_lft forever preferred_lft forever

$ route
Kernel-IP-Routentabelle
Ziel            Router          Genmask         Flags Metric Ref    Use Iface
link-local      *               255.255.0.0     U     1000   0        0 enp0s31f6
192.168.0.0     *               255.255.0.0     U     100    0        0 enp0s31f6
244.244.244.245 *               255.255.255.255 UH    0      0        0 enp0s31f6

$ head ../../config/vsomeip-tcp-client.json 
{
    "unicast" : "192.168.1.1",
    "netmask" : "255.255.0.0",
    "logging" : 
    {
        "level" : "info",
        "console" : "true",
        "file" : { "enable" : "true", "path" : "/var/log/vsomeip.log" },
        "dlt" : "true"
    },

[3] client:
$ vsomeip/build/examples$ env VSOMEIP_CONFIGURATION=../../config/vsomeip-tcp-service.json VSOMEIP_APPLICATION_NAME=service-sample ./response-sample
2018-08-06 17:46:03.896483 [info] Parsed vsomeip configuration in 2ms
2018-08-06 17:46:03.897237 [info] Using configuration file: "../../config/vsomeip-tcp-service.json".
2018-08-06 17:46:03.897319 [info] Default configuration module loaded.
2018-08-06 17:46:03.897349 [info] Initializing vsomeip application "service-sample".
2018-08-06 17:46:03.897508 [info] SOME/IP client identifier configured. Using 1277 (was: 1277)
2018-08-06 17:46:03.897538 [info] Instantiating routing manager [Host].
2018-08-06 17:46:03.897624 [info] init_routing_endpoint Routing endpoint at /tmp/vsomeip-0
2018-08-06 17:46:03.897783 [info] Client [1277] is connecting to [0] at /tmp/vsomeip-0
2018-08-06 17:46:03.897882 [info] Service Discovery enabled. Trying to load module.
2018-08-06 17:46:03.898705 [info] Service Discovery module loaded.
2018-08-06 17:46:03.898859 [info] Application(service-sample, 1277) is initialized (11, 100).
Static routing OFF
2018-08-06 17:46:03.898966 [info] Starting vsomeip application "service-sample" using 2 threads
2018-08-06 17:46:03.899223 [info] shutdown thread id from application: 1277 (service-sample) is: 7f1ba7e9d700 TID: 17438
2018-08-06 17:46:03.899324 [info] Watchdog is disabled!
2018-08-06 17:46:03.899526 [info] io thread id from application: 1277 (service-sample) is: 7f1bab4b5740 TID: 17435
2018-08-06 17:46:03.899583 [info] io thread id from application: 1277 (service-sample) is: 7f1ba6e9b700 TID: 17440
2018-08-06 17:46:03.899773 [info] vSomeIP 2.10.21 | (default)
2018-08-06 17:46:03.899922 [info] Network interface "enx0080c83cef81" state changed: up
2018-08-06 17:46:03.900771 [info] main dispatch thread id from application: 1277 (service-sample) is: 7f1ba869e700 TID: 17437
Application service-sample is registered.
2018-08-06 17:46:03.901006 [info] OFFER(1277): [1234.5678:0.0]
2018-08-06 17:46:03.901394 [info] OFFER(1277): [1235.5678:0.0]
2018-08-06 17:46:13.899908 [info] vSomeIP 2.10.21 | (default)
2018-08-06 17:46:13.903145 [info] STOP OFFER(1277): [1234.5678:0.0]
2018-08-06 17:46:13.903309 [info] STOP OFFER(1277): [1235.5678:0.0]
2018-08-06 17:46:23.900479 [info] vSomeIP 2.10.21 | (default)
2018-08-06 17:46:23.906339 [info] OFFER(1277): [1234.5678:0.0]
2018-08-06 17:46:23.906683 [info] OFFER(1277): [1235.5678:0.0]
…

[4] service:
$ vsomeip/build/examples$ env VSOMEIP_CONFIGURATION=../../config/vsomeip-tcp-client.json VSOMEIP_APPLICATION_NAME=client-sample ./request-sample
2018-08-06 17:52:35.999589 [info] Parsed vsomeip configuration in 0ms
2018-08-06 17:52:35.999686 [info] Using configuration file: "../../config/vsomeip-tcp-client.json".
2018-08-06 17:52:35.999712 [info] Default configuration module loaded.
2018-08-06 17:52:35.999729 [info] Initializing vsomeip application "client-sample".
2018-08-06 17:52:35.999800 [info] SOME/IP client identifier configured. Using 1343 (was: 1343)
2018-08-06 17:52:35.999815 [info] Instantiating routing manager [Host].
2018-08-06 17:52:35.999855 [info] init_routing_endpoint Routing endpoint at /tmp/vsomeip-0
2018-08-06 17:52:35.999928 [info] Client [1343] is connecting to [0] at /tmp/vsomeip-0
2018-08-06 17:52:35.999948 [info] Service Discovery enabled. Trying to load module.
2018-08-06 17:52:36.000151 [info] Service Discovery module loaded.
2018-08-06 17:52:36.000205 [info] Application(client-sample, 1343) is initialized (11, 100).
Client settings [protocol=UDP:quiet=false:cycle=1000]
2018-08-06 17:52:36.000250 [info] Starting vsomeip application "client-sample" using 2 threads
2018-08-06 17:52:36.000368 [info] shutdown thread id from application: 1343 (client-sample) is: 7fa02982d700 TID: 17760
2018-08-06 17:52:36.000375 [info] main dispatch thread id from application: 1343 (client-sample) is: 7fa02a02e700 TID: 17759
2018-08-06 17:52:36.000446 [info] Watchdog is disabled!
Service [1234.5678] is NOT available.
Service [1235.5678] is NOT available.
2018-08-06 17:52:36.000536 [info] io thread id from application: 1343 (client-sample) is: 7fa02ce59740 TID: 17756
2018-08-06 17:52:36.000540 [info] REQUEST(1343): [1234.5678:255.4294967295]
2018-08-06 17:52:36.000551 [info] io thread id from application: 1343 (client-sample) is: 7fa02882b700 TID: 17762
2018-08-06 17:52:36.000658 [info] vSomeIP 2.10.21 | (default)
2018-08-06 17:52:36.000726 [info] Network interface "enp0s31f6" state changed: up
2018-08-06 17:52:46.001025 [info] vSomeIP 2.10.21 | (default)
2018-08-06 17:52:56.001504 [info] vSomeIP 2.10.21 | (default)
2018-08-06 17:53:06.002083 [info] vSomeIP 2.10.21 | (default)
2018-08-06 17:53:16.002653 [info] vSomeIP 2.10.21 | (default)
2018-08-06 17:53:26.003242 [info] vSomeIP 2.10.21 | (default)
…

hemalbavishi, 2018-08-13:
You need to pass --tcp while invoking application on client side.

closed 2018-08-14 by GenivivSOMEIPmaintainer

brenkem, 2018-10-15:
The solution for this problem can also be found here, including a detailed description: https://github.com/GENIVI/vsomeip/issues/36

NandhanThiravia, 2019-02-11:
Hi @brenkem 
I tried https://github.com/GENIVI/vsomeip/issues/36 but I couldn't get over the issue. I have added the code, json files & logs in https://github.com/GENIVI/vsomeip/issues/29#issuecomment-462309685, any help would be really appreciable