COVESA/vsomeip #35: vsomeip tests fail on qemuarm64 target in meta-ivi

frznlogic, 2018-08-21:
Hi, I'm having troubles running vsomeip tests in meta-ivi-test meta layer with the test-image on qemuarm64. It works on both qemux86 and qemux86-64. This is the log output from running the tests:

root@qemuarm64:/opt/tests# ./test-fw.sh vsomeip 2
##########
vSOMEIP
##########
[ run_2 ]
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from someip_big_payload_test
[ RUN      ] someip_big_payload_test.send_ten_messages_to_service
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from someip_big_payload_test
[ RUN      ] someip_big_payload_test.receive_ten_messages_and_send_reply
2018-08-21 10:16:27.378363 [info] Parsed vsomeip configuration in 153ms
2018-08-21 10:16:27.427728 [info] Using configuration file: "big_payload_test_local.json".
2018-08-21 10:16:27.428725 [info] Default configuration module loaded.
2018-08-21 10:16:27.429397 [info] Initializing vsomeip application "big_payload_test_client".
2018-08-21 10:16:27.420495 [info] Parsed vsomeip configuration in 173ms
2018-08-21 10:16:27.445158 [info] Using configuration file: "big_payload_test_local.json".
2018-08-21 10:16:27.446132 [info] Default configuration module loaded.
2018-08-21 10:16:27.446794 [info] Initializing vsomeip application "big_payload_test_service".
2018-08-21 10:16:27.458546 [info] SOME/IP client identifier configured. Using 1344 (was: 1344)
2018-08-21 10:16:27.478102 [info] Instantiating routing manager [Proxy].
2018-08-21 10:16:27.479967 [info] Client [1344] is connecting to [0] at /tmp/vsomeip-0
2018-08-21 10:16:27.489694 [error] Routing manager with id 4727 already exists.
2018-08-21 10:16:27.490707 [error] Couldn't acquire client identifier
/usr/src/debug/vsomeip/2.10.21-r0/git/test/big_payload_tests/big_payload_test_service.cpp:51: Failure
Failed
Couldn't initialize application
2018-08-21 10:16:27.513366 [info] Listening at /tmp/vsomeip-1344
2018-08-21 10:16:27.514339 [info] Application(big_payload_test_client, 1344) is initialized (11, 100).terminate called without an active exception

2018-08-21 10:16:27.554078 [info] Starting...
2018-08-21 10:16:27.565797 [info] Starting vsomeip application "big_payload_test_client" using 2 threads
./big_payload_test_local_starter.sh: line 43:  1524 Aborted                 (core dumped) ./big_payload_test_service $1
2018-08-21 10:16:27.600386 [info] main dispatch thread id from application: 1344 (big_payload_test_client) is: 7f84382140 TID: 1530
2018-08-21 10:16:27.594687 [info] shutdown thread id from application: 1344 (big_payload_test_client) is: 7f83b81140 TID: 1531
2018-08-21 10:16:27.613252 [info] io thread id from application: 1344 (big_payload_test_client) is: 7f84be9010 TID: 1525
2018-08-21 10:16:27.614680 [info] Client 1344 successfully connected to routing  ~> registering..
2018-08-21 10:16:27.625731 [info] io thread id from application: 1344 (big_payload_test_client) is: 7f83380140 TID: 1532
2018-08-21 10:16:28.660003 [warning] Client 0x1344 register timeout! : Restart route to stub!
2018-08-21 10:16:28.665612 [info] Application/Client 1344: Reconnecting to routing manager.
2018-08-21 10:16:28.768359 [info] Client 1344 successfully connected to routing  ~> registering..
2018-08-21 10:16:29.771354 [warning] Client 0x1344 register timeout! : Restart route to stub!
2018-08-21 10:16:29.772984 [info] Application/Client 1344: Reconnecting to routing manager.
2018-08-21 10:16:29.875960 [info] Client 1344 successfully connected to routing  ~> registering..
2018-08-21 10:16:30.879405 [warning] Client 0x1344 register timeout! : Restart route to stub!
2018-08-21 10:16:30.983928 [info] Client 1344 successfully connected to routing  ~> registering..
2018-08-21 10:16:31.987656 [warning] Client 0x1344 register timeout! : Restart route to stub!
2018-08-21 10:16:32.092063 [info] Client 1344 successfully connected to routing  ~> registering..
2018-08-21 10:16:33.095378 [warning] Client 0x1344 register timeout! : Restart route to stub!
2018-08-21 10:16:33.199916 [info] Client 1344 successfully connected to routing  ~> registering..
2018-08-21 10:16:34.203794 [warning] Client 0x1344 register timeout! : Restart route to stub!
2018-08-21 10:16:34.308329 [info] Client 1344 successfully connected to routing  ~> registering..
2018-08-21 10:16:35.311628 [warning] Client 0x1344 register timeout! : Restart route to stub!
2018-08-21 10:16:35.416176 [info] Client 1344 successfully connected to routing  ~> registering..
2018-08-21 10:16:36.419452 [warning] Client 0x1344 register timeout! : Restart route to stub!
2018-08-21 10:16:36.523978 [info] Client 1344 successfully connected to routing  ~> registering..
2018-08-21 10:16:37.527418 [warning] Client 0x1344 register timeout! : Restart route to stub!
2018-08-21 10:16:37.632015 [info] Client 1344 successfully connected to routing  ~> registering..
2018-08-21 10:16:38.635722 [warning] Client 0x1344 register timeout! : Restart route to stub!
2018-08-21 10:16:38.739969 [info] Client 1344 successfully connected to routing  ~> registering..
2018-08-21 10:16:39.743374 [warning] Client 0x1344 register timeout! : Restart route to stub!
2018-08-21 10:16:39.847915 [info] Client 1344 successfully connected to routing  ~> registering..
2018-08-21 10:16:40.851772 [warning] Client 0x1344 register timeout! : Restart route to stub!
2018-08-21 10:16:40.956325 [info] Client 1344 successfully connected to routing  ~> registering..
2018-08-21 10:16:41.959322 [warning] Client 0x1344 register timeout! : Restart route to stub!
2018-08-21 10:16:42.063916 [info] Client 1344 successfully connected to routing  ~> registering..
[... 24 lines cut ...]
2018-08-21 10:16:52.862063 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2018-08-21 10:16:54.472992 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2018-08-21 10:16:54.476383 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2018-08-21 10:16:56.085957 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2018-08-21 10:16:56.090162 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2018-08-21 10:16:57.696131 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2018-08-21 10:16:57.697825 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2018-08-21 10:16:59.305037 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2018-08-21 10:16:59.309586 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2018-08-21 10:17:00.917184 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2018-08-21 10:17:00.922001 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2018-08-21 10:17:02.533047 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2018-08-21 10:17:02.537831 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2018-08-21 10:17:04.145165 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2018-08-21 10:17:04.149849 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
^Croot@qemuarm64:/opt/tests# 2018-08-21 10:17:05.757158 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2018-08-21 10:17:05.761950 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2018-08-21 10:17:07.368173 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2018-08-21 10:17:07.369506 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2018-08-21 10:17:08.977771 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
[... cut ...]
2018-08-21 10:17:10.589174 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2018-08-21 10:17:10.593696 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2018-08-21 10:17:12.201166 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2018-08-21 10:17:12.205587 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED

gunnarx, 2019-01-31:
@martin-ejdestig Let's try to reproduce this and clarify status again. 

The type of errors I see above "Client 0x1344 calling host_->on_state with DEREGISTERED" I have seen mostly when getting some kind of mismatch due to starting/stopping processes.  
On a fresh/clean run I don't think those type of errors should happen.

The tests themselves are written in meta-ivi and I'm not sure how familiar @GenivivSOMEIPmaintainer is with them.  I have a little bit of vSOMEIP experience now and we can try them again together, before we ask for help.

spiderman07, 2019-02-25:
Hi guys. Any update on this? what was the issue?

gunnarx, 2019-02-25:
Not sure if it's component or the integration / tests themselves, but likely the latter because vSOMEIP is used successfully in many areas.  Moving to close this issue is mostly a matter of resource and priority now.    @spiderman07 you might want to try to run the tests in your own environment and report.

spiderman07, 2019-02-25:
@gunnarx i can easily reproduce it. on a clean start works ok. the problem occurs if i run it again but don't power off before(this is a workaround we are using at the moment).

Also i get these logs at the end of the fresh run:
Client [1344] is closing connection to [1277]
local_client_endpoint: connection_reseted/EOF/bad_descriptor
Client [1277] is closing connection to [1344]
routing_manager_proxy::stop unlink failed (/tmp/vsomeip-1277): No such file or directory

Application/Client 1344 is deregistering.
Application/Client 1344 is deregistered.
local_client_endpoint: connection_reseted/EOF/bad_descriptor
routing_manager_proxy::stop unlink failed (/tmp/vsomeip-1344): No such file or directory

Got any suggestion in which direction shall i investigate?

martin-ejdestig, 2019-05-09:
For me the tests fail when running it through https://github.com/GENIVI/meta-ivi/blob/189f536d4e8e0966035afeb0679b1e1dd240d1b7/meta-ivi-test/recipes-core-ivi/vsomeip/vsomeip/vsomeip_t.inc#L21 that pipes `stdout` and `stdin` to a file. They never finish and there is always something like the following at the end of the log:

```
....
 [----------] 1 test from someip_application_exception_test
 [ RUN ] someip_application_exception_test.catch_exception_in_invoked_handler
 2019-05-07 11:30:54.188998 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
 2019-05-07 11:30:54.303820 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
 2019-05-07 11:30:54.509114 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
...
```
If I read the code correctly the error message above comes from: https://github.com/GENIVI/vsomeip/blob/9fb9beecadf52083599302fa8ddee7efbec64a39/implementation/endpoints/src/local_client_endpoint_impl.cpp#L133

The particular test that I pasted failure output from above is here: https://github.com/GENIVI/vsomeip/blob/9fb9beecadf52083599302fa8ddee7efbec64a39/test/application_tests/application_test.cpp#L440

Implemented in the test fixture above: https://github.com/GENIVI/vsomeip/blob/9fb9beecadf52083599302fa8ddee7efbec64a39/test/application_tests/application_test.cpp#L313

If I run the *_starter.sh scripts that are part of the meta-ivi test suite (`application_test_starter.sh`, `big_payload_test_local_starter.sh`, `external_local_payload_test_client_local_starter.sh`, `header_factory_test_send_receive_starter.sh`, `local_payload_test_starter.sh` and `local_routing_test_starter.sh`) manually from within the emulator, they all finish successfully.

Is there some race/timing dependent issue with the tests?... since it does not work when outputting to a file but works when outputting to the terminal (which is slower). I have also never seen a problem when running the tests on my own machine (locally, not in an emulator) or in qemux86 and qemux86-64 emulators.

martin-ejdestig, 2019-05-09:
The above results are with vsomeip currently in meta-ivi which is 2.10.21. That is, current master.

martin-ejdestig, 2019-05-09:
@GenivivSOMEIPmaintainer Any input here? (See above.)

closed 2020-08-11 by lutzbichler

martin-ejdestig, 2020-08-11:
@lutzbichler Nice that the issue is closed. What was done to fix it or conclude that it is not a problem?

lutzbichler, 2020-08-11:
Reason for closing is that we do no longer support v2.