COVESA/vsomeip #38: vsomeip error "libvsomeip-cfg.so.2 "

Azsxzsxa, 2018-09-27:
I try to run the "first application" from the vsomeip git hub page and I get the following error when i do ./service-example : 

[0x00007f2275b62740] [error]   Loading failed: (libvsomeip-cfg.so.2: cannot open shared object file: No such file or directory)
Service Discovery module could not be loaded!

 I'm running it on a VM

Azsxzsxa, 2018-09-28:
i did sudo make install and it didnt appear anymore, thanks tho :)

closed 2018-10-26 by GenivivSOMEIPmaintainer

bit2shift, 2019-07-19:
@Azsxzsxa congratulations, you just polluted your system when there was no need to. Good luck cleaning that mess without a full system reinstall. /s
Assuming you built vsomeip locally, ˋLD_LIBRARY_PATH=<vsomeip-path>/build ./service-exampleˋ is what was suggested. No need to taint your system with stuff you might end up forgetting about in the following weeks.