COVESA/vsomeip #39: Help required on porting the implementation of netlink_connector on QNX

uj864, 2018-10-08:
Hello,

I am trying to port the implementation of vsomeip on QNX. For building the apps I had commented the code of netlink_connector as it was including #include <linux/netlink.h> and #include <linux/rtnetlink.h>. I am able to set up the communication with server running on QNX and client on LINUX but not the other way round. From what I understand the reason is that the client is using the code of netlink_connector to listen to the ethernet port. I will be grateful if somebody can advice on how to do this for QNX.

Thanks

uj864, 2018-10-09:
Closing the issue as the apps work without netlink_connector.

closed 2018-10-09 by uj864

johnsolom, 2020-07-16:
@uj864 I am also facing the same issue can you let me know how you fixed it

robinchung, 2020-10-27:
> Closing the issue as the apps work without netlink_connector.

HI,I met the same issue when I port vsomeip to QNX ,could you plz give some advice how to fix it? just comment the netlink code?