COVESA/vsomeip #40: Connect without service discovery

fmallaba, 2018-10-18:
Is it possible to establish a connection between two Linux machines without service discovery? When I trying use vsomeip with service discovery it works. But I need to statically configure IP addresses. I try to implement one-to-one communication and I know both IP addresses before executing.

GenivivSOMEIPmaintainer, 2018-10-26:
You can use the "static routing" feature. You need to configure the services IP address within the configuration file on client side 

...
"services" :
[
    {
        "service" : "0x1234",
        "instance" : "0x5678",
        "unicast" : "192.168.0.8"
    }
]

and to disable the service discovery.

..
"service-discovery" :
{
    "enable" : "false",
    ....
}

Please note that this only works for services that only expose SOME/IP methods as SOME/IP events/fields require service discovery (as the subscription mechanism is part of the service discovery).

fmallaba, 2018-10-26:
I tried to use 'static routing' feature. It works. But on the client side availability handler triggered right after client application starts, even if the service is not offered on the service side. Can I avoid this behaviour? I want the client to start sending messages only when the service available indeed.

lutzbichler, 2019-07-22:
You cannot avoid this behaviour. Providing service availability is the main purpose of the SOME/IP service discovery. If you switch it off, you need to check it by yourself (e.g. by using a specific request/response method).

closed 2019-07-22 by lutzbichler