COVESA/vsomeip #43: [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)

npucloud, 2019-01-23:
Dear all,
I was trying to make the vsomeip examples run on two devices.
Device1: x86_64 Ubuntu16.04
Device2: ARM Linux
I followed the instructions in /vsomeip/examples/readme.txt.
And I added route with:
route add 224.244.224.245 dev eth0
route add 224.225.226.233 dev eth0
Finally after I started the executable,I got an error and there's no communication succeeded.
【Device1】:
cloud@ubuntu:~/someip/vsomeip/examples/build$ export VSOMEIP_CONFIGURATION=vsomeip-local.json
cloud@ubuntu:~/someip/vsomeip/examples/build$ export VSOMEIP_APPLICATION_NAME=client-sample
cloud@ubuntu:~/someip/vsomeip/examples/build$ ./request-sample
2019-01-23 19:22:27.012182 [info] Parsed vsomeip configuration in 0ms
2019-01-23 19:22:27.012478 [info] Using configuration file: "vsomeip-local.json".
2019-01-23 19:22:27.012524 [info] Default configuration module loaded.
2019-01-23 19:22:27.012545 [info] Initializing vsomeip application "client-sample".
2019-01-23 19:22:27.012967 [info] SOME/IP client identifier configured. Using 1344 (was: 1344)
2019-01-23 19:22:27.013060 [info] Instantiating routing manager [Proxy].
2019-01-23 19:22:27.013457 [info] Client [1344] is connecting to [0] at /tmp/vsomeip-0
2019-01-23 19:22:27.013966 [info] Listening at /tmp/vsomeip-1344
2019-01-23 19:22:27.014002 [info] Application(client-sample, 1344) is initialized (11, 100).
Client settings [protocol=UDP:quiet=false:cycle=1000]
2019-01-23 19:22:27.014092 [info] Starting vsomeip application "client-sample" using 2 threads
2019-01-23 19:22:27.014214 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2019-01-23 19:22:27.014283 [info] io thread id from application: 1344 (client-sample) is: 7f6e4ae8a740 TID: 8278
2019-01-23 19:22:27.014327 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2019-01-23 19:22:27.014513 [info] main dispatch thread id from application: 1344 (client-sample) is: 7f6e482e8700 TID: 8280
2019-01-23 19:22:27.015160 [info] shutdown thread id from application: 1344 (client-sample) is: 7f6e47ae7700 TID: 8281
2019-01-23 19:22:27.015391 [info] io thread id from application: 1344 (client-sample) is: 7f6e472e6700 TID: 8282
2019-01-23 19:22:27.115622 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2019-01-23 19:22:27.115800 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2019-01-23 19:22:27.317420 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2019-01-23 19:22:27.317527 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2019-01-23 19:22:27.719074 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2019-01-23 19:22:27.719418 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2019-01-23 19:22:28.520667 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2019-01-23 19:22:28.520817 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2019-01-23 19:22:30.122305 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2019-01-23 19:22:30.122379 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2019-01-23 19:22:31.724830 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2019-01-23 19:22:31.725134 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
2019-01-23 19:22:33.326825 [warning] local_client_endpoint::connect: Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)
2019-01-23 19:22:33.326977 [info] routing_manager_proxy::on_disconnect: Client 0x1344 calling host_->on_state with DEREGISTERED
^C2019-01-23 19:22:34.596693 [info] Stopping vsomeip application "client-sample".
2019-01-23 19:22:34.599389 [info] utility::auto_configuration_exit: munmap succeeded.

【Device2】:
root@s32v234evb:/mnt/someip/samples# export VSOMEIP_CONFIGURATION=vsomeip-local.json
root@s32v234evb:/mnt/someip/samples# export VSOMEIP_APPLICATION_NAME=service-sample
root@s32v234evb:/mnt/someip/samples# ./response-sample
2018-09-05 03:24:52.368665 [info] Parsed vsomeip configuration in 6ms
2018-09-05 03:24:52.369233 [info] Using configuration file: "vsomeip-local.json".
2018-09-05 03:24:52.369366 [info] Default configuration module loaded.
2018-09-05 03:24:52.369425 [info] Initializing vsomeip application "service-sample".
2018-09-05 03:24:52.369790 [info] SOME/IP client identifier configured. Using 1277 (was: 1277)
2018-09-05 03:24:52.369852 [info] Instantiating routing manager [Host].
2018-09-05 03:24:52.370221 [info] init_routing_endpoint Routing endpoint at /tmp/vsomeip-0
2018-09-05 03:24:52.370473 [info] Client [1277] is connecting to [0] at /tmp/vsomeip-0
2018-09-05 03:24:52.370617 [info] Service Discovery enabled. Trying to load module.
2018-09-05 03:24:52.741474 [info] Service Discovery module loaded.
2018-09-05 03:24:52.741891 [info] Application(service-sample, 1277) is initialized (11, 100).
Static routing OFF
2018-09-05 03:24:52.742157 [info] Starting vsomeip application "service-sample" using 2 threads
2018-09-05 03:24:52.742532 [info] main dispatch thread id from application: 1277 (service-sample) is: 7f9e4f21d0 TID: 993
2018-09-05 03:24:52.742586 [info] shutdown thread id from application: 1277 (service-sample) is: 7f9dcf21d0 TID: 994
2018-09-05 03:24:52.742894 [info] Watchdog is disabled!
Application service-sample is registered.
2018-09-05 03:24:52.743103 [info] io thread id from application: 1277 (service-sample) is: 7f9f68b890 TID: 990
2018-09-05 03:24:52.743260 [info] OFFER(1277): [1234.5678:0.0]
2018-09-05 03:24:52.744192 [info] OFFER(1277): [1235.5678:0.0]
2018-09-05 03:24:52.744193 [info] io thread id from application: 1277 (service-sample) is: 7f9ccf21d0 TID: 996
2018-09-05 03:24:52.749561 [info] vSomeIP 2.10.21 | (default)
2018-09-05 03:24:52.749777 [info] Network interface "eth0" state changed: up
2018-09-05 03:24:52.750064 [info] Route "224.244.224.245/32 if: eth0 gw: n/a" state changed: up
2018-09-05 03:24:52.750436 [info] Port configuration missing for [1235.5678]. Service is internal.
2018-09-05 03:24:52.750557 [info] SOME/IP routing ready.
2018-09-05 03:25:02.745021 [info] STOP OFFER(1277): [1234.5678:0.0]
2018-09-05 03:25:02.745478 [info] STOP OFFER(1277): [1235.5678:0.0]
2018-09-05 03:25:02.749762 [info] vSomeIP 2.10.21 | (default)
2018-09-05 03:25:12.746270 [info] OFFER(1277): [1234.5678:0.0]
2018-09-05 03:25:12.746582 [info] OFFER(1277): [1235.5678:0.0]
2018-09-05 03:25:12.746656 [info] Port configuration missing for [1235.5678]. Service is internal.
2018-09-05 03:25:12.749861 [info] vSomeIP 2.10.21 | (default)

Could anyone please help with this ?
Looking forward to your answer.
Thanks a lot. ^_^
Sincerely Cloud +86 19921955259

kaip-e, 2020-03-10:
@npucloud How did you change the notification file? I wrote a .batch file to set my custom .json for the application on startup:
```
#!/bin/bash
export VSOMEIP_CONFIGURATION=../service.json
export VSOMEIP_APPLICATION_NAME=Service
```
but i still get the same error.

kaip-e, 2020-03-10:
Hey :)
I got the same error you mentioned. As i was able to fix this, i get another one:
```
Controller [1111.2222] NOT available.
2020-03-10 04:30:07.043687 [error] Routing info for remote service could not be found! (0124): [1111.2222.3333] 0001
```
This is my 
Controller.cpp:
```
#include <iomanip>
#include <iostream>
#include <sstream>

#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1111
#define SAMPLE_INSTANCE_ID 0x2222
#define SAMPLE_METHOD_ID 0x3333

std::shared_ptr <vsomeip::application> controller;

void on_message (const std::shared_ptr <vsomeip::message> & _request){

  std::shared_ptr <vsomeip::payload> message_payload = _request -> get_payload();
  vsomeip::length_t message_length = message_payload -> get_length();

  std::stringstream stream;
  for (vsomeip::length_t i = 0; i < message_length; i++){
    stream << std::hex << (int)*(message_payload -> get_data() +i );
  }

  std::cout << "Player message: "
            << stream.str() << std::endl;

  // Response to Player baesd on msg:
  std::shared_ptr <vsomeip::message> controller_response = vsomeip::runtime::get() -> create_response (_request);
  message_payload = vsomeip::runtime::get() -> create_payload();

  std::vector <vsomeip::byte_t> message_payload_data;

  // response = 9 8 7 6 ... 0
  for (int i = 9; i >= 0; i--){
    // pushes i to message_payload_data as byte
    message_payload_data.push_back (i % 256);
  }

  message_payload -> set_data (message_payload_data);
  controller_response -> set_payload (message_payload);

  controller -> send (controller_response, true);
}

int main(){
  controller = vsomeip::runtime::get() -> create_application ("Controller");
  controller -> init();
  controller -> register_message_handler (SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, SAMPLE_METHOD_ID, on_message);
  controller -> offer_service (SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID);
  controller -> start();
}
```
controller.json:
```
{
    "unicast" : "10.103.2.2",
    "netmask" : "255.255.0.0",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
[... 118 lines cut ...]
    },
    "applications" :
    [
        {
            "name" : "Player",
            "id" : "0x0124"
        }
    ],
    "routing" : "Player",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.0.0.251",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "5",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
```

aananthcn, 2020-03-30:
Error "**Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)**" means value for the key (routing) in configuration file vsomeip.json:
... 
"routing" : "**app-name**"
...

and. the argument passed to create_application() call in your application are not matching
...
app = vsomeip::runtime::get()->create_application("**app-name**");
...

tongchiyang, 2023-02-08:
> Error "**Couldn't connect to: /tmp/vsomeip-0 (No such file or directory / 2)**" means value for the key (routing) in configuration file vsomeip.json: ... "routing" : "**app-name**" ...
>
> and. the [...]

This works,nice!Thank you!

vicopei, 2023-03-20:
> This works,nice!Thank you!

老哥,这个到底是啥问题啊?我现在也遇到了,并且还不是必现.我这里测试不能复现,但是同样的json文件,测试那边有时候能复现.

lijingyi2002, 2024-01-17:
> I changed the .json file from vsomeip-local.json to vsomeip.json, problem solved.

I think you may commit a patch to fix the readme issue.

lijingyi2002, 2024-01-17:
> 老哥,这个到底是啥问题啊?我现在也遇到了,并且还不是必现.我这里测试不能复现,但是同样的json文件,测试那边有时候能复现.

应该是这个config文件里面的 routing 对应的内容，
    "routing" : "client-sample",
需要和你环境变量里面的名字对应
export VSOMEIP_APPLICATION_NAME=client-sample