COVESA/vsomeip #44: [error]Routing info for remote service could not be found! (1313): [1234.5678.0421] 0001

npucloud, 2019-01-26:
Dear all,

I followed the Tutorial in this page:
https://github.com/GENIVI/vsomeip/wiki/vsomeip-in-10-minutes#devices
In testing demo [Request/Reponse], I met a problem I cann't solve.
And I've been stuck here for more than three days, so I am sincerely in sore need for any of your help. 
Please contact me if you can help me in any ways.Thanks a lot!
Email: npuliuyun@163.com
TEL: +86 19921955259

I want the demo to run on two devices:
[Device 1]: PC 64bit VMware Ubuntu16.04
[Device 2]: EVB ARM64bit Linux4.13.4
In reference to the Tutorial, I only changed the unicast IP address to 192.168.0.5(service) and 192.168.0.4(client) in .json file. And I do route add 224.224.224.245 dev eth0 on both sides.
 I didn't change anything else but i got the error on client side as shown in title.
Here are the running log: (I started sevice first.)
【Device 2 service log】
export VSOMEIP_CONFIGURATION=rr_service.json
export VSOMEIP_APPLICATION_NAME=service-exmple
./service-example 
2018-09-05 03:27:44.420443 [info] Parsed vsomeip configuration in 4ms
2018-09-05 03:27:44.420992 [info] Using configuration file: "rr_service.json".
2018-09-05 03:27:44.421090 [info] Default configuration module loaded.
2018-09-05 03:27:44.421146 [info] Initializing vsomeip application "World".
2018-09-05 03:27:44.421498 [info] SOME/IP client identifier configured. Using 1212 (was: 1212)
2018-09-05 03:27:44.421556 [info] Instantiating routing manager [Host].
2018-09-05 03:27:44.421852 [info] init_routing_endpoint Routing endpoint at /tmp/vsomeip-0
2018-09-05 03:27:44.422060 [info] Client [1212] is connecting to [0] at /tmp/vsomeip-0
2018-09-05 03:27:44.422138 [info] Service Discovery enabled. Trying to load module.
2018-09-05 03:27:45.034345 [info] Service Discovery module loaded.
2018-09-05 03:27:45.034785 [info] Application(World, 1212) is initialized (11, 100).
2018-09-05 03:27:45.034877 [info] OFFER(1212): [1234.5678:0.0]
2018-09-05 03:27:45.035226 [info] Starting vsomeip application "World" using 2 threads
2018-09-05 03:27:45.035680 [info] shutdown thread id from application: 1212 (World) is: 7fbdf891d0 TID: 997
2018-09-05 03:27:45.035727 [info] main dispatch thread id from application: 1212 (World) is: 7fbe7891d0 TID: 996
2018-09-05 03:27:45.035958 [info] Watchdog is disabled!
2018-09-05 03:27:45.036108 [info] io thread id from application: 1212 (World) is: 7fbf122890 TID: 995
2018-09-05 03:27:45.036163 [info] io thread id from application: 1212 (World) is: 7fbcf891d0 TID: 999
2018-09-05 03:27:45.043158 [info] vSomeIP 2.10.21 | (default)
2018-09-05 03:27:45.043636 [info] Network interface "eth0" state changed: up
2018-09-05 03:27:45.044058 [info] Route "default route (0.0.0.0/0) if: eth0 gw: 192.168.0.1" state changed: up
2018-09-05 03:27:45.044745 [info] SOME/IP routing ready.
2018-09-05 03:27:45.045008 [warning] Route "224.224.224.245/32 if: eth0 gw: n/a" state changed: up
2018-09-05 03:27:55.043298 [info] vSomeIP 2.10.21 | (default)
2018-09-05 03:27:57.229318 [warning] Didn't receive a multicast SD message for 2200ms.
2018-09-05 03:27:59.429635 [warning] Didn't receive a multicast SD message for 2200ms.
2018-09-05 03:28:01.629848 [warning] Didn't receive a multicast SD message for 2200ms.
2018-09-05 03:28:03.830072 [warning] Didn't receive a multicast SD message for 2200ms.
2018-09-05 03:28:05.043451 [info] vSomeIP 2.10.21 | (default)
2018-09-05 03:28:06.030368 [warning] Didn't receive a multicast SD message for 2200ms.
2018-09-05 03:28:08.230573 [warning] Didn't receive a multicast SD message for 2200ms.
2018-09-05 03:28:10.430785 [warning] Didn't receive a multicast SD message for 2200ms.
^C2018-09-05 03:28:12.038404 [info] Stopping vsomeip application "World".
2018-09-05 03:28:12.039255 [info] utility::auto_configuration_exit: munmap succeeded.
2018-09-05 03:28:12.039410 [info] Exiting vsomeip application...

【Device 1 client log】
export VSOMEIP_CONFIGURATION=rr_client.json
export VSOMEIP_APPLICATION_NAME=client-example
./client-example 
2019-01-26 12:27:28.503500 [info] Parsed vsomeip configuration in 13ms
2019-01-26 12:27:28.503774 [info] Using configuration file: "rr_client.json".
2019-01-26 12:27:28.503824 [info] Default configuration module loaded.
2019-01-26 12:27:28.504045 [info] Initializing vsomeip application "Hello".
2019-01-26 12:27:28.504353 [info] SOME/IP client identifier configured. Using 1313 (was: 1313)
2019-01-26 12:27:28.504447 [info] Instantiating routing manager [Host].
2019-01-26 12:27:28.506022 [info] init_routing_endpoint Routing endpoint at /tmp/vsomeip-0
2019-01-26 12:27:28.506386 [info] Client [1313] is connecting to [0] at /tmp/vsomeip-0
2019-01-26 12:27:28.506494 [info] Service Discovery enabled. Trying to load module.
2019-01-26 12:27:28.532966 [info] Service Discovery module loaded.
2019-01-26 12:27:28.533192 [info] Application(Hello, 1313) is initialized (11, 100).
2019-01-26 12:27:28.533348 [info] REQUEST(1313): [1234.5678:255.4294967295]
2019-01-26 12:27:28.533704 [info] Starting vsomeip application "Hello" using 2 threads
2019-01-26 12:27:28.535067 [info] main dispatch thread id from application: 1313 (Hello) is: 7f9b42801700 TID: 2940
2019-01-26 12:27:28.535801 [info] Watchdog is disabled!
2019-01-26 12:27:28.536112 [info] shutdown thread id from application: 1313 (Hello) is: 7f9b42000700 TID: 2941
2019-01-26 12:27:28.536194 [info] io thread id from application: 1313 (Hello) is: 7f9b455fc740 TID: 2938
CLIENT: Service [1234.5678] is NOT available.
2019-01-26 12:27:28.536527 [info] vSomeIP 2.10.21 | (default)
**_2019-01-26 12:27:28.536614 [error] Routing info for remote service could not be found! (1313): [1234.5678.0421] 0001_**
2019-01-26 12:27:28.536753 [info] Network interface "ens33" state changed: up
2019-01-26 12:27:28.536794 [info] io thread id from application: 1313 (Hello) is: 7f9b40ffe700 TID: 2943
2019-01-26 12:27:28.536864 [info] Route "default route (0.0.0.0/0) if: ens33 gw: 192.168.0.1" state changed: up
2019-01-26 12:27:28.540263 [info] SOME/IP routing ready.
2019-01-26 12:27:28.540501 [warning] Route "224.224.224.245/32 if: ens33 gw: n/a" state changed: up
CLIENT: Service [1234.5678] is available.
2019-01-26 12:27:38.536993 [info] vSomeIP 2.10.21 | (default)
2019-01-26 12:27:46.792381 [warning] Didn't receive a multicast SD message for 2200ms.
2019-01-26 12:27:48.537441 [info] vSomeIP 2.10.21 | (default)
2019-01-26 12:27:48.560026 [info] update_routing_info: elapsed=1000 : delete service/instance 1234.5678
CLIENT: Service [1234.5678] is NOT available.
2019-01-26 12:27:49.002506 [warning] Didn't receive a multicast SD message for 2200ms.
2019-01-26 12:27:51.212555 [warning] Didn't receive a multicast SD message for 2200ms.
^C2019-01-26 12:27:52.181039 [info] Stopping vsomeip application "Hello".
2019-01-26 12:27:52.183005 [info] utility::auto_configuration_exit: munmap succeeded.
2019-01-26 12:27:52.183206 [info] Exiting vsomeip application...

npucloud, 2019-01-31:
Is there anyone could help me? Thanks!

npucloud, 2019-02-01:
> I can't guarantee that I have any idea, but try posting your rr_service.json also, for information.

Thanks for your attention. Here are my json files, I only changed the IP address from json info provided in vsomeip in 10 mins tutorial:
[service side: rr_service.json]
{
    "unicast" : "192.168.0.5",
    "logging" :
    { 
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" : 
    [
        {
            "name" : "World",
            "id" : "0x1212"
        }
	{
            "name" : "Hello",
            "id" : "0x1313"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30509"
        }
    ],
    "routing" : "World",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
[client side: rr_client.json]
{
    "unicast" : "192.168.0.4",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/var/log/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" : 
    [
        {
            "name" : "World",
            "id" : "0x1212"
        }
        {
            "name" : "Hello",
            "id" : "0x1313"
        } 
    ],
    "routing" : "Hello",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}

And I also tried the answer in #40, but it didn't work.

lutzbichler, 2019-07-22:
It looks as if the routing is not setup correctly and therefore you do not receive the SOME/IP-SD multicast messages. Try setting:

# route add -net 224.0.0.0/4 dev <YOUR DEVICE>

xuguoxuan, 2019-10-08:
There is a grammatical mistake in your json conf, please try following as:

...
{
"name" : "World",
"id" : "0x1212"
}**,**
{
"name" : "Hello",
"id" : "0x1313"
}
...

sateesh12, 2019-12-19:
I am facing same issue between NVIDIA TX2 and Raspberry Pi 3, which is connected to a switch. Please advise. I have added the route. I get route error only when trying to invoke a method. Service discovery works fine.

ronakvashi, 2020-01-02:
I faced same problem when I tested with two devices connected via switch. Even after adding routing info for multicast IP it didn't work.
The problem is with client-example.cpp given at https://github.com/GENIVI/vsomeip/wiki/vsomeip-in-10-minutes in on_availability handler. It is notifying thread to send request even before service has made request available for clients. This is evident from log message shared in this issue too.

There are two invocation of on_availability, first with _is_available == false and next with true. Original code doesn't check this boolean and just notifies thread to send request via condition variable.

This behavior is not observed when both client and service are on same machine.
Here is modified code.

`void on_availability(vsomeip::service_t _service, vsomeip::instance_t _instance, bool _is_available) {
    std::cout << "CLIENT: Service ["
            << std::setw(4) << std::setfill('0') << std::hex << _service << "." << _instance
            << "] is "
            << (_is_available ? "available." : "NOT available.")
            << std::endl;
    if(_is_available) condition.notify_one();
}`

sateesh12, 2020-01-03:
Thank you rvashi2020, the issue is solved.

pancio88, 2020-01-15:
Hi @sateesh12 , how the issue is solved? I have a similar issue but using CommonAPI library with SomeIP. Same code, different configuration (from UDP to TCP) is not working in the TCP case.

Siddharthav1, 2020-01-23:
Hi @pancio88 sir, Even I am  using CommonAPI library with SomeIP following https://genivi-oss.atlassian.net/wiki/spaces/COMMONAPICPP/pages/5472320/CommonAPI+C+SOME+IP+in+10+minutes , Withnn the system works good. I am trying to communicate bwtewwe two Ubuntu PCs. but the client can detect the presence and absence of the service but the functioncall is not happening.

  Any suggestion please. I have created a new issue also https://github.com/GENIVI/vsomeip/issues/98#issue-552802739 please let me know if anything wrong I am doing here. I have attached all the code that am using here.

pancio88, 2020-01-23:
Hi @Siddharthav1 , I found out that the configuration on vsomeip is not enough to support TCP communication. On our .fdepl file is required to add 'SomeIpReliable = true' to each method that you intend to provide over TCP (See SOMEIP_deployment_spec.fdepl file). Hope will fork for you as well.

Siddharthav1, 2020-01-23:
Actually I am not trying to use TCP I am just trying to test client - server communication in two PCs. Also using a simple helloworld program.

 “Our objective is having two PCs connected With the ethernet cable and achieve inter-PC communication”.
The current setup:

1. 2 PCs with Ubuntu 18.04
2. Common API, VsomeIP, Boost library, SOME/IP has been installed.
3. Configured static IP(locally) for both the PCs.

below is my codes:
[someip.zip](https://github.com/GENIVI/vsomeip/files/4102174/someip.zip)

Any suggestions will help me sir. Even I tried adding  'SomeIpReliable = true' in my case it didn't work.

pancio88, 2020-01-23:
Do you have .pcap capture of the communication between the 2 PCs of the multicast network and that port?
My suggestion is to try also to include also the vsomeip daemon  and try to allow it to be the router for the service discovery network. Not sure if helps.

Siddharthav1, 2020-01-23:
@pancio88 hi sir, Here is my .pcap sir.

[pcapng.zip](https://github.com/GENIVI/vsomeip/files/4102582/pcapng.zip)

pancio88, 2020-01-23:
your destination is not matching with the multicast network in the json file.

try to replace
 $ route add -net 224.0.0.0/4 dev YOUR-DEVICE
with
$ route add -net multicatst-network/32 dev YOUR-DEVICE

Siddharthav1, 2020-01-23:
I tried sir but not working still. 

Here is service PC 
[service.zip](https://github.com/GENIVI/vsomeip/files/4102758/service.zip)
.pcapng after the changes:

pancio88, 2020-01-23:
those pcap the multicast network is 224.0.0.1, the json is "multicast" : "224.224.224.245". Which one you intend to use as multicast network?

Also can you share the console log again

Siddharthav1, 2020-01-23:
hI sir,
below I have attached the recent work:

server:
[service1.zip](https://github.com/GENIVI/vsomeip/files/4103421/service1.zip)

Client:

Src:
[src.zip](https://github.com/GENIVI/vsomeip/files/4103427/src.zip)

Emilyljj, 2020-05-08:
hello, have you found the solution to this problem? I have the same problem. When the transport protocol is UDP , the program works. But when I change the protocol to TCP, there is a "[error] Routing info for remote service could not be found! (5555): [1111.2222.3333] 0001" error.

pancio88, 2020-05-08:
At the end, I solved that when I change to TCP the *.fdepl file which I use to generate the code need to have the following syntax:
`
    method <yourMethod>
    {
        SomeIpMethodID = <Method Id value>
        SomeIpReliable = true (for TCP, or remove this line default is UDP)
    }
`

closed 2020-08-11 by lutzbichler

tiangaomingjing, 2021-10-15:
@Emilyljj  I have the same problem , have you found the solution?

pjacky03, 2022-07-13:
> hello, have you found the solution to this problem? I have the same problem. When the transport protocol is UDP , the program works. But when I change the protocol to TCP, there is a "[error] Routing info for remote service could not be found! (5555): [1111.2222.3333] 0001" error.

I change my configuration file from tcp to udp. It works.

BrunoLSilvaCTW, 2023-10-02:
Hello,

> (...) [error] Routing info for remote service could not be found! (...)

In case anyone is still having this issue with vsomeip, you may need to call [set_reliable(bool)](https://github.com/COVESA/vsomeip/blob/655bb34279bf9ded679451258b8eaf2c8a1a7ea9/interface/vsomeip/message_base.hpp#L184) when creating your request to update the binding that should be used. Use `true` for reliable communication (TCP) or `false` for unreliable (UDP).
```cpp
auto message = runtime->create_request(true);
message->set_service(SERVICE_ID);
message->set_method(METHOD_ID);
message->set_instance(INSTANCE_ID);
```
```cpp
// TCP
message->set_reliable(true);
```
```cpp
// UDP
message->set_reliable(false);
```
Also, make sure that the service is correctly configured to use reliable or unreliable.
```jsonc
// service-tcp.json
"services": [
    {
        "service": "0x1234",
        "instance": "0x5678",
        // TCP
        "reliable": "30501"
    }
],
```
```jsonc
// service-udp.json
"services": [
    {
        "service": "0x1234",
        "instance": "0x5678",
        // UDP
        "unreliable": "30501"
    }
],
```

black-mechanic, 2023-10-02:
> ```c++
> auto message = runtime->create_request(true);
> ```

Hi everyone!
Actualy this is enough for TCP: _auto message = runtime->create_request(**true**);_ when you create object.
So you can skip this call _set_reliable(true/false)_;  It just duplicate the above command, or let you switch protocol on demand, if you need it.

And huge thanks to  @BrunoLSilvaCTW !  👍 
UDP works fine, but TCP!!!  I spent the whole day fighting with json files. I didn't expect such a trick in the library code. :)

maramharsha99, 2023-10-03:
https://github.com/COVESA/vsomeip.git
im trying to execute the hello world programs on TCP and UDP got stuck since 3 to 4 days. Can anyone please help or guide me how to do it?

liuchengloveyou, 2024-01-13:
Hello everyone!
I encountered the same problem and worked on it for two or three days,
As mentioned upstairs, the problem is in creating_ The correct protocol must be specified when requesting

auto rq = rtm_->create_request(**true/*true is TCP,false is UDP*/**);

Because:
std::shared_ptr<message> runtime_impl::create_request(bool _reliable) const {
    std::shared_ptr<message_impl> its_request =
            std::make_shared<message_impl>();
    its_request->set_protocol_version(VSOMEIP_PROTOCOL_VERSION);
    its_request->set_message_type(message_type_e::MT_REQUEST);
    its_request->set_return_code(return_code_e::E_OK);
    its_request->set_reliable(_reliable);
    its_request->set_interface_version(DEFAULT_MAJOR);
    return (its_request);
}

nidheesh-ss, 2024-05-02:
@liuchengloveyou @black-mechanic @BrunoLSilvaCTW  i am also facing the same issues mentioned here as i am trying to communicarte bw two linux PCs i have provided more info inthe request ID please help me
#676