COVESA/vsomeip #48: How to create a Service which offers an event as well as subscribes for another event using VSOMEIP

NandhanThiravia, 2019-03-28:
I created 2 services:
1) **hybrid_service**:
This service would be the routing manager. It would offer an event and also subscribes to an event offered by hybrid_client

2) **hybrid_client**:
This service would NOT be the routing manager. It offers an event which first_service is interested in.

Both the services are able to find each other and even connect but when hybrid_client notifies the hybrid_service with events, hybrid_service is unable to receive the events.

A strange log that gets printed in hybrid_service side is as follows:
```2019-03-28 19:52:06.212659 [info] Notify one event 778 to client 6789 failed. Event payload not set!```

I am attaching the code, config and code files here.
Any help in what wrong I am doing would be appreciated.

[hybrid.zip](https://github.com/GENIVI/vsomeip/files/3018407/hybrid.zip)

NandhanThiravia, 2019-03-28:
Updated the zip file with CMakeList file too.
[hybrid.zip](https://github.com/GENIVI/vsomeip/files/3018425/hybrid.zip)

gunnarx, 2019-07-01:
I have no obvious solution for you but:

In initial testing I have been most successful using a separate routing manager, i.e. "vsomeipd".
So setting in the config.json:   
```"routing" : "vsomeipd" ```
and making sure that binary is running on the local system (with the appropriate config file).  Always double-check that the printouts indicate that it is using the config file you provided and not some "default config".

If it works with vsomeipd well... then at least it might give some information.  If not, then you might focus on the events/notification again.

lutzbichler, 2019-07-18:
You need to call "request_service" for the service containing the event before calling "request_event".

closed 2019-07-30 by lutzbichler