COVESA/vsomeip #49: How to set payload when using offer_event with a none_zero _cycle parameter?Thanks!

npucloud, 2019-04-24:
Dear all,

According to the defination of offer_event in application.hpp:
`    /**
     *
     * \brief Offers a SOME/IP event or field.
     *
     * A user application must call this method for each event/field it wants
     * to offer. The event is registered at the vsomeip routing component that
     * enables other applications to subscribe to the event/field as well as
     * to get and set the field value.
     *
     * This version of offer_event adds some additional functionalities:
     * - It is possible to configure a cycle time. The notification message of
     *   this event is then resent cyclically.
     * - The parameter _change_resets_cycle is available to control how event
     *   notification works in case the data is updated by the application. If
     *   set to true, an update of the data immediately leads to a
     *   notification. Otherwise, the updated data is sent only after the
     *   expiration of the cycle time.
     * - It is possible to specify callback function that can be used to
     *   implement a predicate that determines whether or not two event values
     *   are considered different. Field notifications are only sent if the
     *   predicate evaluates to true (or if a notify method is called with the
     *   force flag being set).
     *
     * \param _service Service identifier of the interface containing the
     * event.
     * \param _instance Instance identifier of the interface containing the
     * event.
     * \param _event Event identifier of the offered event.
     * \param _eventgroups List of eventgroup identifiers of the eventgroups
     * that contain the event.
     * \param _is_field Selector for event or field.
     * \param _cycle Sets the cycle time of the event. If nonzero, data is
     * resent cyclically after the cycle time expired.
     * \param _change_resets_cycle Tells if a change immediately leads to
     * a notification.
     * \param _epsilon_change_func Predicate that determines if two given
     * payloads are considered different.
     *
     * Note: The different versions of offer_event exist for compatibility
     * reasons. They will be merged with the next major vsomeip version.
     */
    virtual void offer_event(service_t _service,
            instance_t _instance, event_t _event,
            const std::set<eventgroup_t> &_eventgroups,
            bool _is_field,
            std::chrono::milliseconds _cycle,
            bool _change_resets_cycle,
            const epsilon_change_func_t &_epsilon_change_func) = 0;`
if I set a none zero _cycle parameter, the notification message would be resent cyclically. So I understanded it as I don't need to call notify manually in this situation. But the notify function need to know the updated data (payload). So if I don't call notify manually,how could the notify function know what the payload is?  Is there any function to call to set the payload for my offer_event?

I checked the source code but only found a function named set_payload which pathetically could not be called from user app. 
Any advice would be appreciated!!! Thanks a lot!!!

npuliuyun@163.com
+86 19921955259

npucloud, 2019-04-30:
Dear all,
Here's my test for offer_event with cycle parameter:
1. call "offer_event" with a cycle parameter:
    app_->offer_event(
                PROVIDED_SERVICE_ID,
                PROVIDED_INSTANCE_ID,
                EVENT_ID_OFFER_FrP02,
                its_groups,
                false,
                std::chrono::milliseconds(CYCLE_25),
                false,
                std::bind(&services::func_FrP02,
                          this,
                          std::placeholders::_1, std::placeholders::_2));

bool services::func_FrP02(const std::shared_ptr<vsomeip::payload> &_payload0,const std::shared_ptr<vsomeip::payload> &_payload1){
        return true;
}

2. call "notify" to set values for event:
payload_FrP02->set_data(data_FrP02);
app_->notify(PROVIDED_SERVICE_ID, PROVIDED_INSTANCE_ID, EVENT_ID_OFFER_FrP02, payload_FrP02);

3. cyclically call "set_data" for updating data of the offered event but this actually didn't work:
payload_FrP02->set_data(data_FrP02);

Then, I got cyclically notification on client side but the payload remained same. So what's the standard way of updating data for an event?
I checked all the APIs in _interface_ folder one by one but didn't find a proper API to update the payload. So I wonder how to update data for event without calling "notify" cyclically in user app? 

And when I changed event to field, I found the client would only get the notification once. It seems the Predicate callback function didn't work. how to make the notification of field to be sent cyclically too? 
"Field notifications are only sent if the predicate evaluates to true."

Sincerely looking forward for your help!
Thanks!
Please contact me if you have any advice ^_^

lutzbichler, 2020-08-11:
Always call notify when you want to change the payload. It depends on the event configuration how the update is handled (immediately sent or with the next cyclic update).

closed 2020-11-27 by lutzbichler