COVESA/vsomeip #51: Unable to build on OSX High Sierra

terrisgit, 2019-05-27:
cmake runs fine but when I run make, I get:

1 error generated.
make[2]: *** [CMakeFiles/vsomeip.dir/implementation/endpoints/src/client_endpoint_impl.cpp.o] Error 1
make[1]: *** [CMakeFiles/vsomeip.dir/all] Error 2
make: *** [all] Error 2

cmake version 3.14.4
gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.0.0 (clang-800.0.38)
Target: x86_64-apple-darwin17.7.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin

terrisgit, 2019-06-02:
Ah. OSX is not supported.

closed 2019-06-02 by terrisgit

cfanatic, 2020-06-29:
You can run vsomeip in a docker container, though. Take a look at https://github.com/cfanatic/vsomeip-fuzzing for reference.