COVESA/vsomeip #52: Vsomeip stack issues

RameshRex, 2019-05-28:
Getting libboost errors while compiling. 
However, is this the full vsomeip stack. How will I test this once i build?

phongt, 2019-06-04:
I can build with boost 1.62.0 
I found 1 patch for  [0001-Support-boost-v1.66.patch](https://github.com/GENIVI/meta-ivi/blob/master/meta-ivi/recipes-core-ivi/vsomeip/vsomeip/0001-Support-boost-v1.66.patch)

RameshRex, 2019-06-04:
Thank you.
The sample Hello world app runs only on one pc. If I run client on a pc and
server on another, it does not communicate.

Any idea what changes need to be made on the json script or any other?

gunnarx, 2019-06-04:
@phongt Agreed, and after applying this patch it works with Boost 1.66 (as stated) and 1.67 also if I remember correctly.   Someone else could try 1.68 / later and report.  Thanks for answering.

@RameshRex:

 "VSomeIP stack issues" is not a good title for a ticket.   I started to rename it for you  "Vsomeip failed to compile with boost <version>" but you did not specify what you had tried version.  Not only in the title but you should have described what you were trying to do in your original question.  This time it was lucky that @phongt could answer, with such little information.

Then, if you are going to quote the text received from @phongt in the comments, select only the text to quote, and then answer below it.  Don't quote text at the bottom, and don't include a big blob of nonsense as done here --> Why are you posting the text " You are receiving this because you authored the thread.... and all the stuff that follows it into the comment section?  Please keep things clear and clean and put a bit more effort into it if you expect others to use their time to help you.

Open one ticket per problem, and describe it clearly.  Don't start discussing a totally different type of problem as you did here in the same place, after getting the answer on the compilation problem.  It should have a new ticket.

Before you add a new ticket for the HelloWorld problem, please read through the closed tickets - others that had problems were given tips on what part of the documentation to read to adjust the configuration files accordingly.   Did you read the documentation and follow those examples?   This Wiki provides useful information, such as the "10 minutes" guide, if you have not used it so far.
https://github.com/GENIVI/capicxx-core-tools/wiki

gunnarx, 2019-06-04:
Just for reference, it seems this ticket is a duplication of this one https://github.com/GENIVI/vsomeip/issues/31

phongt, 2019-06-08:
@RameshRex  
- Maybe what you need [Communication between two devices](https://github.com/GENIVI/vsomeip/wiki/vsomeip-in-10-minutes#communication-between-two-devices) 

- More details doc [here](https://docs.projects.genivi.org/vSomeIP/2.0.1/html/README.html#_configuration_file_structure)

gunnarx, 2019-06-19:
Does this solve your problem @RameshRex and can we close the ticket?

RameshRex, 2019-06-25:
Yes. Thank you.

closed 2019-06-25 by RameshRex

Chandu-dappadi, 2024-03-14:
I have added the boost as submodule and after that i have added vsomeip as submodule but vsomeip is unable to find boost 
getting this error CMake Error in external/vsomeip/CMakeLists.txt:
  Found relative path while evaluating include directories of "vsomeip3":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/CMakeLists.txt:
  Found relative path while evaluating include directories of "vsomeip3-cfg":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/CMakeLists.txt:
  Found relative path while evaluating include directories of "vsomeip3-sd":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/CMakeLists.txt:
  Found relative path while evaluating include directories of "vsomeip3-e2e":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error: install(EXPORT "vsomeip3Targets" ...) includes target "vsomeip3" which requires target "boost_thread" that is not in any export set.
CMake Error: install(EXPORT "vsomeip3Targets" ...) includes target "vsomeip3" which requires target "boost_filesystem" that is not in any export set.
CMake Error in external/vsomeip/examples/routingmanagerd/CMakeLists.txt:
  Found relative path while evaluating include directories of
  "routingmanagerd":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/tools/CMakeLists.txt:
  Found relative path while evaluating include directories of "vsomeip_ctrl":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/examples/CMakeLists.txt:
  Found relative path while evaluating include directories of
  "request-sample":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/examples/CMakeLists.txt:
  Found relative path while evaluating include directories of
  "response-sample":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/examples/CMakeLists.txt:
  Found relative path while evaluating include directories of
  "subscribe-sample":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/examples/CMakeLists.txt:
  Found relative path while evaluating include directories of
  "notify-sample":

    "Boost_INCLUDE_DIR-NOTFOUND"

CMake Error in external/vsomeip/CMakeLists.txt:
  export called with target "vsomeip3" which requires target "boost_thread"
  that is not in any export set.

CMake Error in external/vsomeip/CMakeLists.txt:
  export called with target "vsomeip3" which requires target
  "boost_filesystem" that is not in any export set.

-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    CMAKE_TOOLCHAIN_FILE

CMake Generate step failed.  Build files cannot be regenerated correctly.
make: *** [Makefile:65: all] Error 1