COVESA/vsomeip #53: Client-Sever communication failed

Sabita05, 2019-06-06:
While trying to run the hello-world client -server code(source-https://github.com/GENIVI/vsomeip/tree/master/examples/hello_world) on two different ubuntu pc the communication between the client-server is failing.
Attached the client-server logs and client-server json file for further reference.
I could see in the server log as 
**Port configuration missing for [1111.2222]. Service is internal**
Need to know what should be the ip address and port no. for the client and server.
whether the client will have its own ip in the json file or the server's ip address or vice-versa.
Any help would be appreciated.

[clientjson.txt](https://github.com/GENIVI/vsomeip/files/3260081/clientjson.txt)
[clientlog.txt](https://github.com/GENIVI/vsomeip/files/3260082/clientlog.txt)
[serverjson.txt](https://github.com/GENIVI/vsomeip/files/3260083/serverjson.txt)
[serverlog.txt](https://github.com/GENIVI/vsomeip/files/3260084/serverlog.txt)

phongt, 2019-06-08:
@Sabita05  hope this helpful [vsomeip in 10 minutes](https://github.com/GENIVI/vsomeip/wiki/vsomeip-in-10-minutes#devices)

closed 2019-06-13 by Sabita05

bularcasergiu, 2021-01-15:
This issue is reproduced using vSomeIP 3.1.7.1, and affects the service side when on the target is running a routing manager:
```
 [info] OFFER(4444): [1111.2222:0.0] (true)
 [info] Port configuration missing for [1111.2222]. Service is internal.
```
If i configure my service application also as routing manager the service is registered correctly.
[service_app_as_routing_manager.txt](https://github.com/GENIVI/vsomeip/files/5821960/service_app_as_routing_manager.txt)
[service_app_log_with_default_routing.txt](https://github.com/GENIVI/vsomeip/files/5821961/service_app_log_with_default_routing.txt)
[service_running_with_default_routing_manager_conf.txt](https://github.com/GENIVI/vsomeip/files/5821962/service_running_with_default_routing_manager_conf.txt)
[routing_manager_conf.txt](https://github.com/GENIVI/vsomeip/files/5821963/routing_manager_conf.txt)
[routing_manager_log.txt](https://github.com/GENIVI/vsomeip/files/5821964/routing_manager_log.txt)
[service_app_as_routing_manager_conf.txt](https://github.com/GENIVI/vsomeip/files/5821965/service_app_as_routing_manager_conf.txt)

I need to start a routing manager because on my targets are running more than one vsomeip application. 
These service needs to communicate with a client deployed on other target.