COVESA/vsomeip #54: crypto support

DevNaga, 2019-06-07:
Hello GENIVI team,

This issue is only a -know how- about crypto offering in SOME/IP. I went through the Autosar SOME/IP spec and however failed to see the underlying crypto and configuration support to enable this one.

It looks like the above CommonAPI layer hides in the underlying serialization and transport method and for the applications using the CommonAPI the communication will simply look like a function call rather an actual serialize -> send -> receive -> deserailize call. So there is no way knowing the encoded wire-data that is being communicated. 

Specifically i am looking for:

1. crypto algorithm to be used / key lengths
2. cert storage path / configuration
3. configuration in choosing the authenticate vs encryption (symmetric and asymmetric)

I am  not really sure if this support already is present in this version of vsomeip here. If possible could you please point me towards the link / source file /configuration about such implementation ?

Thanks,
Dev

DevNaga, 2019-06-07:
More so, what is the underlying key exchange protocol ? How to distribute (be in symmetric or asymmetric) keys between one publisher and multiple subscribers ? should this be a job of the vsomeipd or publisher should do this as another thread instance ?

lutzbichler, 2019-07-22:
There is no encryption support in vsomeip. The only encoding that is supported by vsomeip/CommonAPI is the SOME/IP encoding which is not encrypted.

closed 2019-07-22 by lutzbichler

DevNaga, 2019-07-22:
Thank you. Is there any way the encryption could be part of new version of SOME/IP Autosar Specification ?

gunnarx, 2019-07-23:
That is a question you would have to ask in a AUTOSAR forum, since details of not yet released specifications are only discussed among AUTOSAR members.

(I would guess currently it is assumed that the comm channel is protected, if needed. E.g. by applying SSL/TLS)

kkretsch, 2021-07-30:
If unencrypted multicasts through untrusted networks are still state of the art, where to complain? I think an additional layer of at least payload obfuscation is strongly advised. Any links to Autosar in the web? Or other upcoming projects for this?