COVESA/vsomeip #55: Does vsomeip have plans to support Android?

AnaBit, 2019-06-19:
I want to import vsomeip into the android system, but I have encountered many problems, such as shared memory functions are not supported. Does vsomeip have plans to support android system?

gunnarx, 2019-06-19:
Yes, this is an interesting question for many companies, and investigations are ongoing.  If you start trying it, please share your experiences (with more detail).  Do you get a compilation error for the shared memory functions?

AnaBit, 2019-06-21:
Yes, I have encountered two problems, one is shared memory, one is std::dynamic_pointer_cast The return value is NULL, this should be a problem with NDK. https://github.com/android-ndk/ndk/issues/533; https://github.com/android-ndk/ndk/issues/519

gunnarx, 2019-07-11:
Thanks for your input @AnaBit.
GENIVI alliance is running a special-interest-group (SIG) for automotive companies that are working with Android.  I think it's likely that we use that group to try to synchronize different initiatives around the ability to use SOME-IP in systems that also include an Android ECU.  And I think that would start relatively soon.   Hopefully you can participate in this discussion.

lutzbichler, 2019-07-19:
Android support is planned for vsomeip 3.

AnaBit, 2019-07-22:
When is the plan to release vsomeip3.0?

lutzbichler, 2019-07-24:
It is planned to be released until end of this month.

closed 2019-07-25 by AnaBit

zhyh2010, 2019-10-12:
Excuse me, it's mentioned that vsomeip3 should support android ECU at the end of 2019.7.30 , where can I find it?

zeerd, 2019-10-21:
it there any news for this v3?

zeerd, 2019-11-25:
@gunnarx  I am interesting in this SIGN, any way to join it?

nkh-lab, 2020-06-30:
Now vsomeip supports both AOSP and NDK builds, here is examples:
https://github.com/GENIVI/vsomeip/blob/master/examples/hello_world/readme_android