COVESA/vsomeip #56: vSOMEIP example does not work correctly.

AnaBit, 2019-07-02:
I'm running vSOMEIP example in the diff devices with ubuntu os. two devices connected by LAN.
### devices A vSOMEIP service:
**example running by cmd :** 
`env VSOMEIP_CONFIGURATION=../../config/vsomeip-local-tcp-service.json VSOMEIP_APPLICATION_NAME=service-sample ./response-sample`
running log info : 
app name : service-sample
2019-07-01 20:31:21.944028 [info] Parsed vsomeip configuration in 0ms
2019-07-01 20:31:21.944132 [info] Using configuration file: "../../config/vsomeip-tcp-service.json".
2019-07-01 20:31:21.944183 [info] Default configuration module loaded.
2019-07-01 20:31:21.944210 [info] Initializing vsomeip application "service-sample".
2019-07-01 20:31:21.944358 [info] SOME/IP client identifier configured. Using 1277 (was: 1277)
2019-07-01 20:31:21.944385 [info] Instantiating routing manager [Host].
2019-07-01 20:31:21.944462 [info] init_routing_endpoint Routing endpoint at /tmp/vsomeip-0
2019-07-01 20:31:21.944577 [info] Client [1277] is connecting to [0] at /tmp/vsomeip-0
2019-07-01 20:31:21.944612 [info] Service Discovery enabled. Trying to load module.
2019-07-01 20:31:21.944887 [info] Service Discovery module loaded.
2019-07-01 20:31:21.945006 [info] Application(service-sample, 1277) is initialized (11, 100).
Static routing OFF
2019-07-01 20:31:21.945073 [info] Starting vsomeip application "service-sample" using 2 threads
2019-07-01 20:31:21.945279 [info] shutdown thread id from application: 1277 (service-sample) is: 7fdf2b418700 TID: 4123
2019-07-01 20:31:21.945371 [info] Watchdog is disabled!
2019-07-01 20:31:21.945307 [info] main dispatch thread id from application: 1277 (service-sample) is: 7fdf2bc19700 TID: 4122
Application service-sample is registered.
2019-07-01 20:31:21.945551 [info] OFFER(1277): [1234.5678:0.0]
2019-07-01 20:31:21.945560 [info] io thread id from application: 1277 (service-sample) is: 7fdf2c4d0e40 TID: 4120
2019-07-01 20:31:21.945600 [info] io thread id from application: 1277 (service-sample) is: 7fdf2a416700 TID: 4125
2019-07-01 20:31:21.945684 [info] vSomeIP 2.10.21 | (default)
2019-07-01 20:31:21.945749 [info] OFFER(1277): [1235.5678:0.0]
2019-07-01 20:31:21.945859 [info] Network interface "enp2s0" state changed: up
2019-07-01 20:31:21.945990 [info] Route "default route (0.0.0.0/0) if: enp2s0 gw: 192.168.5.30" state changed: up
2019-07-01 20:31:21.946231 [info] SOME/IP routing ready.
2019-07-01 20:31:21.946304 [warning] Route "224.244.224.245/32 if: enp2s0 gw: n/a" state changed: up
2019-07-01 20:31:26.505856 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-01 20:31:28.706029 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-01 20:31:30.906159 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-01 20:31:31.945799 [info] vSomeIP 2.10.21 | (default)
2019-07-01 20:31:31.946643 [info] STOP OFFER(1277): [1234.5678:0.0]
2019-07-01 20:31:31.946837 [info] STOP OFFER(1277): [1235.5678:0.0]
2019-07-01 20:31:33.106299 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-01 20:31:35.306428 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-01 20:31:37.506561 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-01 20:31:39.706693 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-01 20:31:41.906838 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-01 20:31:41.945929 [info] vSomeIP 2.10.21 | (default)
2019-07-01 20:31:41.947766 [info] OFFER(1277): [1234.5678:0.0]
2019-07-01 20:31:41.947919 [info] OFFER(1277): [1235.5678:0.0]
2019-07-01 20:31:44.106970 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-01 20:31:46.307099 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-01 20:31:48.507234 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-01 20:31:50.707361 [warning] Didn't receive a multicast SD message for 2200ms.
^C2019-07-01 20:31:51.016388 [info] Stopping vsomeip application "service-sample".
2019-07-01 20:31:51.017109 [info] utility::auto_configuration_exit: munmap succeeded.
2019-07-01 20:31:51.017178 [info] Exiting vsomeip application...
###  devices B vSOMEIP client:
**example running by cmd**
`env VSOMEIP_CONFIGURATION=../../config/vsomeip-local-tcp-client.json VSOMEIP_APPLICATION_NAME=client-sample ./request-sample`
running log info : 
app name : client-sample
2019-07-02 08:31:32.083397 [info] Parsed vsomeip configuration in 0ms
2019-07-02 08:31:32.083475 [info] Using configuration file: "../../config/vsomeip-tcp-client.json".
2019-07-02 08:31:32.083488 [info] Default configuration module loaded.
2019-07-02 08:31:32.083546 [info] Initializing vsomeip application "client-sample".
2019-07-02 08:31:32.083607 [info] SOME/IP client identifier configured. Using 1343 (was: 1343)
2019-07-02 08:31:32.083629 [info] Instantiating routing manager [Host].
2019-07-02 08:31:32.083659 [info] init_routing_endpoint Routing endpoint at /tmp/vsomeip-0
2019-07-02 08:31:32.083721 [info] Client [1343] is connecting to [0] at /tmp/vsomeip-0
2019-07-02 08:31:32.083736 [info] Service Discovery enabled. Trying to load module.
2019-07-02 08:31:32.083857 [info] Service Discovery module loaded.
2019-07-02 08:31:32.083903 [info] Application(client-sample, 1343) is initialized (11, 100).
Client settings [protocol=UDP:quiet=false:cycle=1000]
2019-07-02 08:31:32.083924 [info] Starting vsomeip application "client-sample" using 2 threads
2019-07-02 08:31:32.084068 [info] main dispatch thread id from application: 1343 (client-sample) is: 7fbbfbdd5700 TID: 6126
2019-07-02 08:31:32.084224 [info] shutdown thread id from application: 1343 (client-sample) is: 7fbbfb5d4700 TID: 6127
2019-07-02 08:31:32.084277 [info] Watchdog is disabled!
Service [1234.5678] is NOT available.
Service [1235.5678] is NOT available.
2019-07-02 08:31:32.084393 [info] io thread id from application: 1343 (client-sample) is: 7fbbfc684800 TID: 6124
2019-07-02 08:31:32.084417 [info] REQUEST(1343): [1234.5678:255.4294967295]
2019-07-02 08:31:32.084457 [info] io thread id from application: 1343 (client-sample) is: 7fbbfa5d2700 TID: 6129
2019-07-02 08:31:32.084639 [info] vSomeIP 2.10.21 | (default)
2019-07-02 08:31:32.084672 [info] Sent READY to systemd watchdog
2019-07-02 08:31:32.084715 [info] Network interface "eth1" state changed: up
2019-07-02 08:31:32.084750 [info] Route "default route (0.0.0.0/0) if: eth1 gw: 192.168.5.30" state changed: up
2019-07-02 08:31:32.084875 [info] SOME/IP routing ready.
2019-07-02 08:31:32.084934 [warning] Route "224.244.224.245/32 if: eth1 gw: n/a" state changed: up
Service [1234.5678] is available.
2019-07-02 08:31:32.113485 [error] Routing info for remote service could not be found! (1343): [1234.5678.0421] 0001
Client/Session [1343/0001] sent a request to Service [1234.5678]
Service [1235.5678] is available.
2019-07-02 08:31:34.995823 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-02 08:31:39.754184 [warning] tcp_client_endpoint receive_cbk: End of file(2) local: 192.168.5.30:41234 remote: 192.168.5.20:30509
2019-07-02 08:31:39.754381 [warning] tcp_client_endpoint receive_cbk restarting.
Service [1234.5678] is NOT available.
2019-07-02 08:31:39.754768 [warning] tce::restart: local:  remote: 192.168.5.20:30509
Service [1235.5678] is NOT available.
2019-07-02 08:31:41.955513 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-02 08:31:42.085020 [info] vSomeIP 2.10.21 | (default)
2019-07-02 08:31:44.157430 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-02 08:31:46.357795 [warning] Didn't receive a multicast SD message for 2200ms.
2019-07-02 08:31:48.558917 [warning] Didn't receive a multicast SD message for 2200ms.
Service [1235.5678] is available.
Service [1234.5678] is available.
2019-07-02 08:31:49.797453 [error] Routing info for remote service could not be found! (1343): [1234.5678.0421] 0001
Client/Session [1343/0001] sent a request to Service [1234.5678]
2019-07-02 08:31:52.085135 [info] vSomeIP 2.10.21 | (default)
^C2019-07-02 08:31:54.177018 [info] Stopping vsomeip application "client-sample".
2019-07-02 08:31:54.177318 [info] Sent STOPPING to systemd watchdog
2019-07-02 08:31:54.177613 [info] utility::auto_configuration_exit: munmap succeeded.
2019-07-02 08:31:54.177643 [info] Exiting vsomeip application...

### device B client running error with this message: 
2019-07-02 08:31:49.797453 [error] Routing info for remote service could not be found! (1343): [1234.5678.0421] 0001
And have no request send to service

Routing information has been added：
`sudo route add -host 224.244.224.245 dev eth1`
`sudo route add -net 224.0.0.0/4 dev eth1`

closed 2019-07-05 by AnaBit

kruemelbacke, 2019-11-12:
Is there a solution?

dcs2020, 2022-07-21:
Hi all,
I am also facing the same issue service not available. Please let me know how you solved it.

hanmhs9, 2023-03-15:
Same issue. UDP works but TCP doesnt work. it says routing info cannot be found but still it sends the request, receiving no reply. can someone help?

BrunoLSilvaCTW, 2023-10-02:
Hello,

> (...) [error] Routing info for remote service could not be found! (...)

In case anyone is still having this issue with vsomeip, you may need to call [set_reliable(bool)](https://github.com/COVESA/vsomeip/blob/655bb34279bf9ded679451258b8eaf2c8a1a7ea9/interface/vsomeip/message_base.hpp#L184) when creating your request to update the binding that should be used. Use `true` for reliable communication (TCP) or `false` for unreliable (UDP).
```cpp
auto message = runtime->create_request(true);
message->set_service(SERVICE_ID);
message->set_method(METHOD_ID);
message->set_instance(INSTANCE_ID);
```
```cpp
// TCP
message->set_reliable(true);
```
```cpp
// UDP
message->set_reliable(false);
```
Also, make sure that the service is correctly configured to use reliable or unreliable.
```jsonc
// service-tcp.json
"services": [
    {
        "service": "0x1234",
        "instance": "0x5678",
        // TCP
        "reliable": "30501"
    }
],
```
```jsonc
// service-udp.json
"services": [
    {
        "service": "0x1234",
        "instance": "0x5678",
        // UDP
        "unreliable": "30501"
    }
],
```