COVESA/vsomeip #57: How does the server pass structures to the client ?

xuguoxuan, 2019-07-06:
I learned "vsomeip in 10 minutes" and can run hello world sever/client on my device. But sever and client comunite with string data, I don't know how to pass structures, that's my real purpose. Could anyone tell me how to implement that on vsomeip? Thans very much!

gunnarx, 2019-07-08:
As you may have seen in the the "payload" related APIs in vSOMEIP, they deliver a byte stream.  Not strings necessarily, but an unstructured array of bytes.   You can therefore choose yourself how to serialize complex data into the payload sent with vSOMEIP, if you wanted to.   

However the SOME/IP specifications from AUTOSAR also specify how structs should be serialized, so that different implementations will be compatible.  Specifications are published on the AUTOSAR web site - for example [This ZIP file](https://www.autosar.org/fileadmin/Releases_TEMP/Foundation_1.5.1/Protocols.zip) includes the SOME/IP protocol specification.

A lot of development with SOME/IP and complex will use a higher level API/binding also for programming.  For example https://github.com/GENIVI/capicxx-someip-runtime (CommonAPI) or ARA:COM for AUTOSAR.   When using these frameworks, the interfaces and necessary datatypes would also be specified in an abstract interface description language, such as Franca IDL, or AUTOSAR XML.  The CommonAPI framework follows the SOME/IP specifications for encoding datatypes.

gunnarx, 2019-07-11:
Does this answer your question @xuguoxuan ?

xuguoxuan, 2019-07-19:
> Does this answer your question @xuguoxuan ?

I'm sorry that it's been so long time can't response to you since I was very busy...But your answer was very nice and my question got  resolved. Thanks very much! @

closed 2019-07-19 by xuguoxuan