COVESA/vsomeip #58: vsomeip communication between docker containers on linux

Prasad454, 2019-07-25:
Hello,

I have created 2 docker containers as below:
[image]

And i am able to communicate between two applications using vsomeip as mentioned in the vsomeip example.

But i want communicate with vsomeip between the two containers, i explored through online but i dont find any useful info.

I am very new to linux platform.

Kindly help me in providing the possibilities for my requirement.

AnaBit, 2019-07-30:
Are the IP addresses of two dockers in the same network segment?

lutzbichler, 2019-07-30:
In general communication between (docker) containers is no different from communication between regular devices. Thus, you can use the configuration examples from the config directory (adapting the IP addresses configured as "unicast" parameter and setting the multicast route should be all that is necessary). For any more target oriented help please provide more detailed information about your configuration, what you are trying to do/achieve and what is going wrong.

closed 2019-09-24 by lutzbichler