COVESA/vsomeip #61: Should we keep version in a separe file?

namsanjp, 2019-08-08:
I think we should have a version file seperately:
`#The version is in format 'Major.Minor.Patch.Hotfix' 
#Hotfix is an optional
#VERSION=2.10.21
VERSION=2.14.a.b`
And CMakeLists.txt should parse it as. It would be easier to count and manage the version number:
`# Copyright (C) 2015-2017 Bayerische Motoren Werke Aktiengesellschaft (BMW AG)
# This Source Code Form is subject to the terms of the Mozilla Public
# License, v. 2.0. If a copy of the MPL was not distributed with this
# file, You can obtain one at http://mozilla.org/MPL/2.0/.

cmake_minimum_required (VERSION 2.8.12)
project (vsomeip)

file(STRINGS "${CMAKE_CURRENT_SOURCE_DIR}/VERSION" VERSION_STRING REGEX "^VERSION")
string(REGEX MATCHALL "[0-9]+" VERSION_LIST ${VERSION_STRING})
list(LENGTH VERSION_LIST VERSION_LENGTH)
if(${VERSION_LENGTH} LESS 3)
	message(STATUS "Missing or invalid VERSION, should in format: 'VERSION=x.x.x or VERSION=x.x.x.x'")
	return()
endif()
list(GET VERSION_LIST 0 VSOMEIP_MAJOR_VERSION)
list(GET VERSION_LIST 1 VSOMEIP_MINOR_VERSION)
list(GET VERSION_LIST 2 VSOMEIP_PATCH_VERSION)
if(${VERSION_LENGTH} GREATER 3)
	list(GET VERSION_LIST 3 VSOMEIP_HOTFIX_VERSION)
else()
	set(VSOMEIP_HOTFIX_VERSION 0)
endif()

message(STATUS "VERSION: ${VSOMEIP_MAJOR_VERSION}.${VSOMEIP_MINOR_VERSION}.${VSOMEIP_PATCH_VERSION}.${VSOMEIP_HOTFIX_VERSION}" )`

lutzbichler, 2019-08-19:
Could you explain the benefit from introducing this. Why do you need to count version numbers?

namsanjp, 2019-08-19:
When writing an external tool need this version number, it can read the same vsomeip version from the 'VERSION' file. So it will be consistent  in the whole project, defined in one place and used by multiple tools. My case, I created a tool to write other doc, and the 'VERSION' file is more convenient.

joao-d, 2023-04-14:
This is implementation is not expected for now, but feel free to add this idea https://github.com/COVESA/vsomeip/discussions/categories/ideas to be taken into account for the future. Thanks

closed 2023-04-14 by goncaloalmeida