COVESA/vsomeip #63: Build process failed on linux mint and ubunut

LamdaLamdaLamda, 2019-08-25:
Build process fails at **[100%] Linking CXX executable vsomeipd**. Throws the following error: **CMakeFiles/vsomeipd.dir/vsomeipd.cpp.o: In function `std::thread::_State_impl<std::thread::_Invoker<std::tuple<vsomeipd_process(bool)::{lambda()#1}> > >::_M_run()':
/usr/local/include/boost/log/sources/basic_logger.hpp:456: undefined reference to `boost::log::v2_mt_posix::core::get_logging_enabled() const' ...**

Boost 1.65 is currently installed but on 1.55 the same error was thrown. Boost itself is perfectly functional. I´ve tested it on Ubuntu and also Mint. None of it worked properly (same error was thrown). A few month ago the vsomeIP worked properly on Ubuntu. Maybe someone has an idea?

lutzbichler, 2019-09-24:
Which ubuntu version are you using and what is the output of "apt list --installed | grep boost" ?

ilyashamd, 2020-01-06:
@LamdaLamdaLamda Did you figure a solution? I am having the same issue.

lutzbichler, 2020-08-11:
boost::log is no longer used.

closed 2020-08-11 by lutzbichler