COVESA/vsomeip #64: SOMEIP TestStub Enhanced Testability Service (ETS)

xuguoxuan, 2019-08-28:
Could you please give some idea of SOMEIP ETS? It satisfies the "OPEN Alliance Automotive Ethernet ECU Test Specification" . Thank you very much!

xuguoxuan, 2020-04-02:
can anyone please share some experience for this issue?

deansay, 2020-07-10:
I already have passed ETS 3.0/2.0 with vsomeip3. QQ276564084

splywaczyk, 2021-01-19:
> I already have passed ETS 3.0/2.0 with vsomeip3. QQ276564084

Could you tell us if you had written your own ETS? Was it based directly on **vsomeip** library or maybe some other like **CommonAPI**?

deansay, 2021-01-20:
it's based directly on vsomeip library.

splywaczyk, 2021-01-25:
@deansay - did you implement ETS by yourself?
I'm trying to do it but wondering for example how parse a payload for `triggerEventUINT8`. There are three arguments which are passed to this method but there is no type so actually it is not known if for example `start` has 1,2 or 4 bytes.

deansay, 2021-01-29:
1) The three arguments data type defined as uint32 which are delay, duration, denounce time. 
2) Then you need to follow Figure 8: triggerEventUINT8Reliable(), that is the same thing.

for example.
```
void service_ETS::thread_EventUINT8(uint32_t start, uint32_t duration, uint32_t debounce)
{
    log(zDebug, "start");
    std::this_thread::sleep_for(std::chrono::seconds(start));
    log(zDebug, "sleep for start:%d",start);
    uint32_t current = 0;
    while(current < duration){
        std::this_thread::sleep_for(std::chrono::seconds(debounce));
        log(zDebug, "sleep for debounce:%d",debounce);
        current += debounce;
        std::shared_ptr<vsomeip::payload> its_payload
            = vsomeip::runtime::get()->create_payload();
        std::vector<vsomeip::byte_t> its_payload_data;
        set_uint8(its_payload_data, current);
        its_payload->set_data(its_payload_data);
        app->notify(ETS_SERVICE_ID0, ETS_INSTANCE_ID1, TestEventUINT8, its_payload);
        log(zDebug, "notify current:%d", current);
    }
    log(zDebug, "end");
}

void service_ETS::on_triggerEventUINT8(const std::shared_ptr<message> &_response)
{
    log(zDebug, "start");

    uint32_t start    = get_uint32(_response->get_payload()->get_data());
    uint32_t duration = get_uint32(&_response->get_payload()->get_data()[4]);
    uint32_t debounce = get_uint32(&_response->get_payload()->get_data()[8]);

    log(zDebug, "start: %d duration: %d debounce: %d", start, duration, debounce);

    if(_response->get_message_type() != message_type_e::MT_REQUEST_NO_RETURN){
        log(zDebug, "this message is not a F&F requset.");
        return;
    }
    log(zDebug, "create thread for EventUINT8");
    std::thread start_thread(std::bind(&service_ETS::thread_EventUINT8, this, start, duration, debounce));
    start_thread.detach();

    log(zDebug, "end");
}
```

splywaczyk, 2021-01-29:
@deansay - this is what I expected. Thanks for the clarification :)

yunfeiyangcc, 2021-04-20:
@deansay @splywaczyk

Hello, What functions need to be implemented in ETS?

How to implement the interface 'suspendInterface' in ETS service?

Do you have any suggestions or sample code?

I hope you can help me. Thank you very much.

splywaczyk, 2021-04-20:
> Hello, What functions need to be implemented in ETS?

All defined in **OPEN Alliance Automotive Ethernet ECU Test Specification - TC8 ECU Test** specification

> How to implement the interface 'suspendInterface' in ETS service?

Specification says that:
_This method allows to suspend the Enhanced Testability Service; thus, forcing the DUT to stop offering the service and start to reoffer the service after a given time._
It takes two parameters:

- start - wait time before the service should be stopped
- duration - wait time before the service should be offered again

so in pseudocode it could look like this:
```
void suspendInterface(uint start, uint duration)
{
    sleep(start);
    stopOfferService();
    sleep(duration);
    startOfferService();
}
```

LccGitHub, 2022-02-21:
@deansay @splywaczyk

Hello,  my test for  [SOMEIPSRV_SD_MESSAGE_07] case in TC8 is always failed。

Do you have any suggestions or vsomeip.json for  pass this case?

I hope you can help me. Thank you very much.

[image]

closed 2023-05-26 by goncaloalmeida

SwanYing, 2023-08-29:
could you give some examples for ets cases，i want to learn this project

bfzravi, 2024-02-15:
Hi,
Kindly request you to share the SOMEIP TestStub Enhanced Testability Service (ETS) cases for learning!...

Thanks