COVESA/vsomeip #74: Facing linker error

arunprasath31, 2019-10-10:
While building I'm facing a linker error. **for vsomeipd**

----------------------------------------------------------
../libvsomeip.so.2.14.16: undefined reference to `virtual thunk to vsomeip::client_endpoint_impl<boost::asio::ip::udp>::set_established(bool)'
../libvsomeip.so.2.14.16: undefined reference to `non-virtual thunk to vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::get_remote_address(boost::asio::ip::address&) const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::connect_cbk(boost::system::error_code const&)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::send_to(std::shared_ptr<vsomeip::endpoint_definition>, unsigned char const*, unsigned int, bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::shutdown_and_close_socket_unlocked(bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::send(std::vector<unsigned char, std::allocator<unsigned char> > const&, unsigned char const*, unsigned int, bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::~client_endpoint_impl()'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::get_remote_port() const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::set_established(bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::get_local_port() const'
../libvsomeip.so.2.14.16: undefined reference to `virtual thunk to vsomeip::client_endpoint_impl<boost::asio::ip::udp>::stop()'
../libvsomeip.so.2.14.16: undefined reference to `virtual thunk to vsomeip::client_endpoint_impl<boost::asio::ip::udp>::is_established() const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::send(std::vector<unsigned char, std::allocator<unsigned char> > const&, unsigned char const*, unsigned int, bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::check_packetizer_space(unsigned int)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::start_connect_timer()'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::connect_cbk(boost::system::error_code const&)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::shutdown_and_close_socket(bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::start_connect_timer()'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::get_remote_port() const'
../libvsomeip.so.2.14.16: undefined reference to `non-virtual thunk to vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::get_remote_port() const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::send_cbk(boost::system::error_code const&, unsigned long, std::shared_ptr<std::vector<unsigned char, std::allocator<unsigned char> > >)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::set_connected(bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::send_cbk(boost::system::error_code const&, unsigned long, std::shared_ptr<std::vector<unsigned char, std::allocator<unsigned char> > >)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::get_remote_address(boost::asio::ip::address&) const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::get_remote_address(boost::asio::ip::address&) const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::send(unsigned char const*, unsigned int, bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::get_remote_address(boost::asio::ip::address&) const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::stop()'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::get_remote_port() const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::send(std::vector<unsigned char, std::allocator<unsigned char> > const&, unsigned char const*, unsigned int, bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::send(unsigned char const*, unsigned int, bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::send(unsigned char const*, unsigned int, bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::set_established(bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::set_connected(bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::is_established() const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::send_or_start_flush_timer(bool, bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::shutdown_and_close_socket_unlocked(bool)'
../libvsomeip.so.2.14.16: undefined reference to `non-virtual thunk to vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::get_remote_address(boost::asio::ip::address&) const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::~client_endpoint_impl()'
../libvsomeip.so.2.14.16: undefined reference to `non-virtual thunk to vsomeip::client_endpoint_impl<boost::asio::ip::udp>::get_remote_address(boost::asio::ip::address&) const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::client_endpoint_impl(std::shared_ptr<vsomeip::endpoint_host>, boost::asio::ip::basic_endpoint<boost::asio::ip::udp>, boost::asio::ip::basic_endpoint<boost::asio::ip::udp>, boost::asio::io_context&, unsigned int, unsigned int)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::send_to(std::shared_ptr<vsomeip::endpoint_definition>, unsigned char const*, unsigned int, bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::check_queue_limit(unsigned char const*, unsigned int) const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::is_established() const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::shutdown_and_close_socket(bool)'
../libvsomeip.so.2.14.16: undefined reference to `non-virtual thunk to vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::get_remote_port() const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::udp>::is_established() const'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::ip::tcp>::set_established(bool)'
../libvsomeip.so.2.14.16: undefined reference to `virtual thunk to vsomeip::client_endpoint_impl<boost::asio::ip::udp>::send_to(std::shared_ptr<vsomeip::endpoint_definition>, unsigned char const*, unsigned int, bool)'
../libvsomeip.so.2.14.16: undefined reference to `virtual thunk to vsomeip::client_endpoint_impl<boost::asio::ip::udp>::send(std::vector<unsigned char, std::allocator<unsigned char> > const&, unsigned char const*, unsigned int, bool)'
../libvsomeip.so.2.14.16: undefined reference to `vsomeip::client_endpoint_impl<boost::asio::local::stream_protocol>::stop()'
[... cut ...]
daemon/CMakeFiles/vsomeipd.dir/build.make:105: recipe for target 'daemon/vsomeipd' failed
make[2]: *** [daemon/vsomeipd] Error 1
CMakeFiles/Makefile2:359: recipe for target 'daemon/CMakeFiles/vsomeipd.dir/all' failed
make[1]: *** [daemon/CMakeFiles/vsomeipd.dir/all] Error 2
Makefile:129: recipe for target 'all' failed
make: *** [all] Error 2

----------------------------------------------------------

**Installed boost 1.66, gcc 7.4 and whatever required as per the documentation. 
Using Ubuntu 18.04.**

closed 2019-10-15 by arunprasath31

SirCosty, 2023-05-17:
Wait another 5 years and maybe will get solved

fcmonteiro, 2023-05-17:
Are you having this issue with vSomeIP 3.3?

RaniaCH0URA, 2024-02-22:
> Wait another 5 years and maybe will get solved

i think so