COVESA/vsomeip #78: Win x64  VS 15 2017, vcpkg, CMake 3.14 Builds has Linker error

TheDraguun, 2019-10-16:
If I disable the emptying of the Boost_LIBRARIES variable it just works...
```
    #set(Boost_LIBRARIES "")
    link_directories(${Boost_LIBRARY_DIR_DEBUG})
```
What is the reason for emptying that variable?

energystoryhhl, 2019-10-18:
boost1.65 msvc14.2 use cmake compile is ok

TheDraguun, 2019-10-24:
cmake compile might not trigger the code block containing the offending line.

What interests me most, is the reason for emptying that variable....

energystoryhhl, 2019-10-24:
you can add boost lib and include dir to system path,the cmake will detecte it

TheDraguun, 2019-10-26:
Thanks for your input, but I have no problem building vsomeip - I just have to disable this in the build script: `set(Boost_LIBRARIES "")` 
I am wondering if maybe there is something wrong with the build script or if I it intentionally is targeting a very specific build environment.

mahaase, 2020-08-12:
same issue here. i am also just wondering about this line...
IMO that should never a good idea to manipulate the boost variables...

lutzbichler, 2020-12-04:
Fixed in v3.1.20.1

closed 2020-12-04 by lutzbichler