COVESA/vsomeip #80: Port 65535 is not an illiegal port - preventing subscription

BjoernAtBosch, 2019-10-21:
Dear vsomeip developers!

In file interface/vsomeip/constants.hpp port number 65535 is defined as constant ILLEGAL_PORT.

But 65535 **is a legal** port number (see https://en.wikipedia.org/wiki/List_of_TCP_and_UDP_port_numbers or https://www.iana.org/assignments/service-names-port-numbers/service-names-port-numbers.xml).

In my case this definition of vsomeip implementation prevents an event subscription via Service Discovery protocol: QNX OS seems using "ad-hoc" port numbers starting with 65535 and then counting down.

In the vsomeip.log at server side I can see the log entry:
`2019-10-21 09:01:36.179957 [warning] Subscription for [000a.5ac0.000b] not valid: Event configuration does not match the provided endpoint options: 192.168.100.21:65535 0.0.0.0:65535 192.168.100.21 session: 0001`

This originates from function service_discovery_impl::handle_eventgroup_subscription, which does the check for "illegal port". This behaviour seems being introduced with version 2.10.11

This should be corrected. The only reserved port number is 0. So, the implementation should either not check for illegal ports at all or maybe use 0 for this purpose.

Thanks for taking care and best regards,
Björn

juliandm, 2020-03-01:
Any news on that? I assume it is related to #111

lutzbichler, 2020-12-04:
Fixed with v3.1.20.2

closed 2020-12-04 by lutzbichler