COVESA/vsomeip #81: vsomeip does not send SUBSCRIBE after service is restart

hobooom, 2019-10-22:
Hi,

I am testing vsomeip with notify-sample and subscribe-sample.
- notify-sample uses vsomeip-udp-service.json
- subscribe-sample uses vsomeip-udp-client.json

When two programs are launched in any order, subscribe-sample sends "SUBSCRIBE" to notify-sample. And "NOTIFICATION" messages are also sent from notify-sample.
If I stop and start again notify-sample, the subscribe-sample does not send "SUBSCRIBE".

If I modify service_discovery_impl.cpp:1429 as below, It works.

        remove_remote_offer_type(its_service, its_instance,
                                 (its_reliable_port != ILLEGAL_PORT ?
                                         its_reliable_address : its_unreliable_address));
        //unsubscribe_all(its_service, its_instance);                     <==== I changed this
        if (!is_diagnosis_ && !is_suspended_) {
            host_->del_routing_info(its_service, its_instance,
                                    (its_reliable_port != ILLEGAL_PORT),
                                    (its_unreliable_port != ILLEGAL_PORT));
        }

Is it BUG?

tjpsantiago, 2019-11-25:
hi @hobooom ,

I think your fix is incorrect because if you check the wrapping if, else-if statement of the line that you've changed, you will notice that the block that you've changed is the behavior for STOP OFFER Service Entry (OFFER SERVICE messages with ttl = 0).

I think this is a problem with the incorrect handling of ttl_timer timeout on the routing manager which triggers the update on the routing info. What you can try to solve this problem is change the if statement under vsomeip::routing_manager_impl::update_routing_info(std::chrono::milliseconds _elapsed)

from
    `if (precise_ttl.count() < _elapsed.count() || precise_ttl.count() == 0)`
to
    `if (precise_ttl.count() <= _elapsed.count() || precise_ttl.count() == 0)`

lutzbichler, 2020-08-11:
Is this local or IP communication?

closed 2020-11-10 by lutzbichler