COVESA/vsomeip #86: vsomeip communication between two devices

Revertigo, 2019-11-11:
i tried to run the simple example from 
https://github.com/GENIVI/vsomeip/wiki/vsomeip-in-10-minutes#prep 
between two different machines.
**machine one #1 - Ubuntu 18.04 - i used the following code(Machine IP 192.168.0.12)**

#include <vsomeip/vsomeip.hpp>
#include <iostream>
#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678

std::shared_ptr< vsomeip::application > app;

int main() {

    app = vsomeip::runtime::get()->create_application("World");
    app->init();
    app->offer_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID);
    app->start();
}
**and i used the following configuration file: vsomeip-world.json**

{
    "unicast" : "192.168.0.12",
    "logging" :
    { 
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" : 
    [
        {
            "name" : "World",
            "id" : "0x1212"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30509"
        }
    ],
    "routing" : "World",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}

**I also configured routing:**
route add -nv 224.224.224.245 dev enp0s3
**Command:**
VSOMEIP_APPLICATION_NAME=World ./APP_NAME

**machine one #2 - Elinos - i used the following code(Machine IP 192.168.0.3)** 
#include <iomanip>
#include <iostream>

#include <vsomeip/vsomeip.hpp>

#define SAMPLE_SERVICE_ID 0x1234
#define SAMPLE_INSTANCE_ID 0x5678

std::shared_ptr< vsomeip::application > app;

void on_availability(vsomeip::service_t _service, vsomeip::instance_t _instance, bool _is_available) {
    std::cout << "Service ["
            << std::setw(4) << std::setfill('0') << std::hex << _service << "." << _instance
            << "] is " << (_is_available ? "available." : "NOT available.")  << std::endl;
}

int main() {

    app = vsomeip::runtime::get()->create_application("Hello");
    app->init();
    app->register_availability_handler(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID, on_availability);
    app->request_service(SAMPLE_SERVICE_ID, SAMPLE_INSTANCE_ID);
    app->start();
}

**configuration file:**
{
    "unicast" : "192.168.0.3",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/var/log/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" : 
    [
        {
            "name" : "Hello",
            "id" : "0x1313"
        } 
    ],
    "routing" : "Hello",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}

**Routing:**
route add -nv 224.224.224.245 dev eth0
**Command:**
VSOMEIP_APPLICATION_NAME=Hello ./APP_NAME

I checked Wireshark on Ubuntu and i can see packets sent to 224.224.224.245:30490

Output from Ubuntu:
2019-11-11 09:40:55.731673 [info] Parsed vsomeip configuration in 0ms
2019-11-11 09:40:55.731727 [info] Using configuration file: "../../../config/vsomeip-world.json".
2019-11-11 09:40:55.731742 [info] Default configuration module loaded.
2019-11-11 09:40:55.731751 [info] Initializing vsomeip application "World".
2019-11-11 09:40:55.731798 [info] SOME/IP client identifier configured. Using 1212 (was: 1212)
2019-11-11 09:40:55.731811 [info] Instantiating routing manager [Host].
2019-11-11 09:40:55.731833 [info] init_routing_endpoint Routing endpoint at /tmp/vsomeip-0
2019-11-11 09:40:55.731879 [info] Client [1212] is connecting to [0] at /tmp/vsomeip-0
2019-11-11 09:40:55.731900 [info] Service Discovery enabled. Trying to load module.
2019-11-11 09:40:55.732002 [info] Service Discovery module loaded.
2019-11-11 09:40:55.732038 [info] Application(World, 1212) is initialized (11, 100).
2019-11-11 09:40:55.732049 [info] OFFER(1212): [1234.5678:0.0]
After offering service
2019-11-11 09:40:55.732112 [info] Starting vsomeip application "World" using 2 threads
2019-11-11 09:40:55.732232 [info] main dispatch thread id from application: 1212 (World) is: 7fcecdcff700 TID: 2363
2019-11-11 09:40:55.732414 [info] Watchdog is disabled!
2019-11-11 09:40:55.732454 [info] shutdown thread id from application: 1212 (World) is: 7fcecd4fe700 TID: 2364
2019-11-11 09:40:55.732497 [info] io thread id from application: 1212 (World) is: 7fced03dfe40 TID: 2362
2019-11-11 09:40:55.732543 [info] vSomeIP 2.10.21 | (default)
2019-11-11 09:40:55.732542 [info] io thread id from application: 1212 (World) is: 7fcec7fff700 TID: 2366
2019-11-11 09:40:55.732611 [info] Network interface "enp0s3" state changed: up
2019-11-11 09:40:55.732670 [info] Route "224.0.0.0/4 if: enp0s3 gw: n/a" state changed: up
2019-11-11 09:40:55.733286 [info] SOME/IP routing ready.

output from Elinos:
2019-11-11 00:16:29.754711 [info] Parsed vsomeip configuration in 5ms
2019-11-11 00:16:29.755602 [info] Using configuration file: "vsomeip-hello.json".
2019-11-11 00:16:29.755823 [info] Default configuration module loaded.
2019-11-11 00:16:29.755964 [info] Initializing vsomeip application "Hello".
2019-11-11 00:16:29.756649 [warning] Could _not_ initialize auto-configuration: Cannot guarantee unique application identifiers!
2019-11-11 00:16:29.756790 [info] Instantiating routing manager [Host].
2019-11-11 00:16:29.757256 [info] init_routing_endpoint Routing endpoint at /tmp/vsomeip-0
2019-11-11 00:16:29.757662 [info] Client [1313] is connecting to [0] at /tmp/vsomeip-0
2019-11-11 00:16:29.757897 [info] Service Discovery enabled. Trying to load module.
2019-11-11 00:16:29.759751 [info] Service Discovery module loaded.
2019-11-11 00:16:29.760317 [info] Application(Hello, 1313) is initialized (11, 100).
2019-11-11 00:16:29.760540 [info] REQUEST(1313): [1234.5678:255.4294967295]
2019-11-11 00:16:29.761321 [info] Starting vsomeip application "Hello" using 2 threads
2019-11-11 00:16:29.762538 [info] Watchdog is disabled!
2019-11-11 00:16:29.763022 [info] io thread id from application: 1313 (Hello) is: 77f5c020 TID: 93
2019-11-11 00:16:29.763391 [info] vSomeIP 2.10.21 | (default)
2019-11-11 00:16:29.763798 [info] io thread id from application: 1313 (Hello) is: 75ed5450 TID: 97
2019-11-11 00:16:29.764514 [info] shutdown thread id from application: 1313 (Hello) is: 76ed5450 TID: 95
2019-11-11 00:16:29.764927 [info] main dispatch thread id from application: 1313 (Hello) is: 776d5450 TID: 94
**Service [1234.5678] is NOT available.**
2019-11-11 00:16:39.771206 [info] vSomeIP 2.10.21 | (default)

I tried to run also Request-Response example and it didn't work as well.
I check other issues having similar problems to mine but it didn't help also.
I'm using on both machines same version 2.10.21 Tag version.
two machine has connectivity(i checked ping).
Can anyone suggest anything i should check or do ?

zeerd, 2019-11-13:
Seems sth wrong in you vsomeip-hello.json?

2019-11-11 00:16:29.756649 [warning] Could not initialize auto-configuration: Cannot guarantee unique application identifiers!

Revertigo, 2019-11-14:
I replaced the Elinos machine with another Ubuntu machine, configured it to be the client(using the same configuration file vsomeip-hello.json, same routing command, same execution command) and it worked as expected. It didn't output:
 "[warning] Could not initialize auto-configuration: Cannot guarantee unique application identifiers!".

I tried to explore the code which triggers this warning but i didn't understand match. 
Can anyone tell what might be the reason for that warning?
I also configured multicast communication within the Elinos kernel.

Revertigo, 2019-11-20:
So after exploring the code once more, i noticed a function shm_open() inside utility.cpp. 
This function isn't supported by my Elinos version so i had to edit the code to use regular open() instead in order to allocate the shared memory(the call to mmap() stays the same. After overcome this issue the module outputted a single warning message which i also seen on regular Ubuntu machine(the code runs without errors on that Ubuntu so i guess it shouldn't be a  problem on Elinos):
1970-01-01 00:03:52.761696 [warning] Routing Manager seems to be inactive. Taking over...

Unfortunately, It's still not working and now i really don't know what to do to get this work properly.

tjpsantiago, 2019-11-25:
have you tried adding clients config on your machine #1 to set what port the client will use to communicate with the service?

```
{
    "unicast" : "192.168.0.12",
    "logging" :
    {
        "level" : "debug",
        "console" : "true",
        "file" : { "enable" : "false", "path" : "/tmp/vsomeip.log" },
        "dlt" : "false"
    },
    "applications" :
    [
        {
            "name" : "World",
            "id" : "0x1212"
        }
    ],
    "services" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30509"
        }
    ],
    **"clients" :
    [
        {
            "service" : "0x1234",
            "instance" : "0x5678",
            "unreliable" : "30509"
        }
    ],**
    "routing" : "World",
    "service-discovery" :
    {
        "enable" : "true",
        "multicast" : "224.224.224.245",
        "port" : "30490",
        "protocol" : "udp",
        "initial_delay_min" : "10",
        "initial_delay_max" : "100",
        "repetitions_base_delay" : "200",
        "repetitions_max" : "3",
        "ttl" : "3",
        "cyclic_offer_delay" : "2000",
        "request_response_delay" : "1500"
    }
}
```

What you could also do is check the traces/packets between the two machines via a packet sniffer like wireshark to see if machine#2 is receiving the offer services from machine#1.

You can also check if machine#2 is already joined the multicast group where service discovery of machine#1 sends its service discovery packets.

lutzbichler, 2020-03-24:
From the information above, the shm is not your problem (as you have configured client identifier as well as routing manager). It looks, as if the multicasts are not reaching the application on the Elinos side. Did you ensure that the route is correctly configured to accept the multicasts?

closed 2020-08-11 by lutzbichler