COVESA/vsomeip #90: [BUG] Routing Info not updated properly due to improper timeout handling

tjpsantiago, 2019-11-22:
Hi,

while using vsomeip, CommonAPI, and CommonAPI SomeIP Binding on our work, we found a bug in the code. TTL timeout checking in service discovery is not properly handled.

Under vsomeip::routing_manager_impl::update_routing_info function, when the function was searching for expired offers, the check doesn't consider **precise_ttl.count() == _elapsed.count()** as a timeout/expired offer.

this causes our notification from one device to another (connected via ethernet) not push through due to port unreachable after disconnect and reconnect of ethernet cable. Routing Info is not updated properly after reconnection of cable even if offer should be considered as expired already.

tjpsantiago, 2019-12-10:
https://github.com/GENIVI/vsomeip/blob/ee44d19e0a3773adcda3bace073c66e334a3e6a7/implementation/routing/src/routing_manager_impl.cpp#L2697

shouldn't it be `if (precise_ttl.count() <= _elapsed.count() || precise_ttl.count() == 0) {` since if it is equal it means that it already timed out right?

lutzbichler, 2020-08-11:
If it is equal we consider the update to be in time. Only if the elapsed time is greater than the ttl we want to expire.

closed 2020-08-11 by lutzbichler