COVESA/vsomeip #93: services with unreliable(UDP) not working

navneet01, 2019-11-29:
i have below service - 
it works with reliable (TCP) , But its not working with unreliable (UDP)

    "services" :
    [
        {
            "service" : "0x04",
            "instance" : "0x01",
            "reliable" : "51004"
        }
    ],

    "services" :
    [
        {
            "service" : "0x04",
            "instance" : "0x01",
            "unreliable" : "51004"
        }
    ],

hobooom, 2019-11-29:
Check client side.
The client should create request message with 'unreliable'.

> runtime.hpp
> virtual std::shared_ptr<message> create_request( bool _reliable = false) const = 0;

navneet01, 2019-11-29:
> Check client side.
> The client should create request message with 'unreliable'.
> 
> > runtime.hpp
> > virtual std::shared_ptr create_request( bool _reliable = false) const = 0;

i have tried this solution in client side  , but no success.

lutzbichler, 2020-01-28:
For sure not general problem. Please provide code sample to show what you are doing to send the messages.

closed 2020-03-24 by lutzbichler

kkretsch, 2021-08-12:
Same problem still here. I created two clients (udp and tcp -> unreliable and reliable) but it seems to always use the TCP connection, regardless of the create_request parameter.