COVESA/vsomeip #97: Unable to execute subscribe_notify_one_test case

arcsweety, 2020-01-03:
Hi,

I am using vsomeip 2.10.10 version. Trying to execute subscribe_notify_one_* test cases between two ubuntu Pcs.

1st Ubuntu Pc:
./subscribe_notify_one_test_master_starter.sh UDP subscribe_notify_one_test_diff_client_ids_diff_ports_master.json 

2nd Ubuntu Pc:
./subscribe_notify_one_test_slave_starter.sh UDP subscribe_notify_one_test_diff_client_ids_diff_ports_slave.json 

Test case does not get exited gracefully.
On console, I see error as "Subscribe to legacy selective service" inspite of setting "supports_selective_broadcasts".

I have set value of supports_selective_broadcasts as "160.160.160.160" on both the PC.
Also tried setting different values as "192.168.0.1" on one Pc and "192.168.0.3" on another Pc.

I have also tested with the latest 2.14.16 version, but no change in the output.
Need suggestions to execute the test case.
Thanks in advance.

lutzbichler, 2020-01-28:
Which IP addresses are you using and how did you call cmake (especially how did you specifiy TEST_IP_MASTER and TEST_IP_SLAVE)?

kaip-e, 2020-03-13:
Did you build googletest 1.7 before using the tests?
If you did maybe:
a) GTEST_ROOT is not found while you call cmake to build vsomeip (Can you doublecheck?)
b) Did you use anotehr version of gtest ? @arcsweety

lutzbichler, 2020-08-11:
No reply. Closing.

closed 2020-08-11 by lutzbichler