COVESA/vsomeip #100: Crash: Threading issue on shutdown?

ManuelKugelmann, 2020-01-30:
I repeatedly got a crash on shutdown. I think it could be a threading issue when remove_timer is working on the linked list of timers but have not investigated in depth.
`
Stack Trace of Crashed Thread 21936:
0x00007FFEA68A8BF7 (vsomeip-sd) [d:\vcpkg\installed\x64-windows\include\boost\asio\detail\timer_queue.hpp:290] boost::asio::detail::timer_queue<boost::asio::detail::chrono_time_traits<std::chrono::steady_clock,boost::asio::wait_traits<std::chrono::steady_clock> > >::remove_timer 
0x00007FFEA6890855 (vsomeip-sd) [d:\vcpkg\installed\x64-windows\include\boost\asio\detail\timer_queue.hpp:193] boost::asio::detail::timer_queue<boost::asio::detail::chrono_time_traits<std::chrono::steady_clock,boost::asio::wait_traits<std::chrono::steady_clock> > >::cancel_timer 
0x00007FFEA681F606 (vsomeip-sd) [d:\vcpkg\installed\x64-windows\include\boost\asio\detail\impl\select_reactor.hpp:71] boost::asio::detail::select_reactor::cancel_timer<boost::asio::detail::chrono_time_traits<std::chrono::steady_clock,boost::asio::wait_traits<std::chrono::steady_clock> > > 
0x00007FFEA68906F1 (vsomeip-sd) [d:\vcpkg\installed\x64-windows\include\boost\asio\detail\deadline_timer_service.hpp:144] boost::asio::detail::deadline_timer_service<boost::asio::detail::chrono_time_traits<std::chrono::steady_clock,boost::asio::wait_traits<std::chrono::steady_clock> > >::cancel 
0x00007FFEA68905A1 (vsomeip-sd) [d:\vcpkg\installed\x64-windows\include\boost\asio\basic_waitable_timer.hpp:361] boost::asio::basic_waitable_timer<std::chrono::steady_clock,boost::asio::wait_traits<std::chrono::steady_clock> >::cancel 
0x00007FFEA67CF22B (vsomeip-sd) [d:\someip\vsomeip-unity\vsomeip\implementation\service_discovery\src\service_discovery_impl.cpp:2702] vsomeip::sd::service_discovery_impl::stop_ttl_timer 
0x00007FFEA67B86AD (vsomeip-sd) [d:\someip\vsomeip-unity\vsomeip\implementation\service_discovery\src\service_discovery_impl.cpp:178] vsomeip::sd::service_discovery_impl::stop 
0x00007FFEA70FE8AA (vsomeip) [d:\someip\vsomeip-unity\vsomeip\implementation\routing\src\routing_manager_impl.cpp:252] vsomeip::routing_manager_impl::stop 
0x00007FFEA736BD88 (vsomeip) [d:\someip\vsomeip-unity\vsomeip\implementation\runtime\src\application_impl.cpp:2019] vsomeip::application_impl::shutdown `

ManuelKugelmann, 2020-01-30:
Related?
https://stackoverflow.com/questions/43168199/cancelling-boost-asio-deadline-timer-safely
https://stackoverflow.com/questions/55200310/can-asio-timer-cancel-call-a-spurious-success

goncaloalmeida, 2023-03-15:
@ManuelKugelmann this is still occurring? Could you please confirm if this is occurring on version 3.3.0 (master branch)?

closed 2023-03-28 by goncaloalmeida