COVESA/vsomeip #104: MSVC vsomeip3 with VSOMEIP_ENABLE_MULTIPLE_ROUTING_MANAGERS linking fails

ManuelKugelmann, 2020-02-18:
`2>configuration_plugin_impl.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: virtual __cdecl vsomeip_v3::plugin::~plugin(void)" referenced in function "int `public: __cdecl vsomeip_v3::plugin_impl<class vsomeip_v3::configuration_plugin_impl>::plugin_impl<...
2>configuration_plugin_impl.obj : error LNK2019: unresolved external symbol "__declspec(dllimport) public: __cdecl vsomeip_v3::plugin::plugin(void)" referenced in function "public: __cdecl vsomeip_v3::plugin_impl<class ...
2>D:\SomeIP\vsomeip-bin\debug\vsomeip3.dll : fatal error LNK1120: 2 unresolved externals
`
I guess this case is not covered by 
`
#if _WIN32
    #if VSOMEIP_DLL_COMPILATION_PLUGIN
        #define VSOMEIP_IMPORT_EXPORT_PLUGIN __declspec(dllexport)
    #else
        #define VSOMEIP_IMPORT_EXPORT_PLUGIN __declspec(dllimport)
    #endif
...
#endif
 
class VSOMEIP_IMPORT_EXPORT_PLUGIN plugin { ...
`

ManuelKugelmann, 2020-02-18:
Excluding "configuration_plugin_impl.cpp" from the main build solves the problem.

lutzbichler, 2020-06-02:
Solved with v3.1.14.1

closed 2020-06-02 by lutzbichler