COVESA/vsomeip #108: How can a host offer two different services on two different ip addresses in one system?

xuguoxuan, 2020-02-27:
Can someone could give some suggestion.
I want to offer two different services, service A and service B. Service A's IP address is 192.168.0.1, but service B's IP address is 192.168.1.1. When service A offered, service B was internal, that bothered me very much .
So, when I want to implement the above case, how the json file should be ?

ManuelKugelmann, 2020-02-27:
I think via configuration overlay.

From the docs (in the applications section of the configuration):
```
** 'overlay' (optional)
+
Contains the path to a configuration that overwrites specific configuration elements
(unicast, netmask, device, network, diagnosis address & mask, service discovery) for the 
application. This allows to manage different network addresses from a single process.
+
NOTE: This feature is only available if vsomeip was compiled with ENABLE_CONFIGURATION_OVERLAYS.
+
```

xuguoxuan, 2020-02-28:
> I think via configuration overlay.
>
> From the docs (in the applications section of the configuration):
>
> ```
> ** 'overlay' (optional)
> +
> Contains the path to a configuration that overwrites specific [...]

Thanks very much for your answer. But OVERLAY is a new funtion in VSOMEIP3, when I use an older vsomeip stack, how can I implement that also ?

ManuelKugelmann, 2020-03-12:
VSOMEIP3 has a compatibility mode so that you can build new libraries that you can use in place of the old ones. Haven't tested that myself yet ...

closed 2020-03-24 by lutzbichler