COVESA/vsomeip #114: overlays can not work at the same time that multiple routing manager is working

xuguoxuan, 2020-03-04:
I found a interesting phenomenon, like the title says. Is there any reason for that? Can anyone please explain that?

ManuelKugelmann, 2020-03-12:
Explain some more ...

xuguoxuan, 2020-03-12:
> Explain some more ...

For example, I want to init two application in one process, and I want to these two both becomes host routing manager on different IP address, we can app A with unicast "192.168.1.1"and app B with unicast "192.168.2.1", so that situation , I compiled vsomeip3 libs with  ENABLE_CONFIGURATION_OVERLAYS and VSOMEIP_ENABLE_MULTIPLE_ROUTING_MANAGERS, and my process program was ran depend on those libs, but when process's runing, I saw overlays json file was not read by my process successfully.
Base on the situation above, I consult the source code, like the picture following...
"vsomeip-master/implementation/application_impl.cpp"
[image]
So, can you explain that is a bug?

ManuelKugelmann, 2020-03-12:
I tinkered with configurations ab bit as well - afaik `configuration_path` does not get in initialized. I exposed it by making it a parameter of the `init`  method. Then you can pass in different config files.

I think the next step is defining different networks (maybe cannot communicate with each other? ... have not tested yet) for the different routing managers. See user guide: 
`
* 'network'
+
Network identifier used to support multiple routing managers on one host. This
setting changes the name of the shared memory segment in `/dev/shm` and the name
of the unix domain sockets in `/tmp/`. Defaults to `vsomeip` meaning the shared
memory will be named `/dev/shm/vsomeip` and the unix domain sockets will be
named `/tmp/vsomeip-$CLIENTID`
`
[https://github.com/GENIVI/vsomeip/blob/master/documentation/vsomeipUserGuide](url)

Why do you need multiple routing managers? Maybe you're on the wrong track... I tried this up to the `network` part and then saw that it was the wrong thing for what I wanted to do....

xuguoxuan, 2020-03-16:
> I tinkered with configurations ab bit as well - afaik `configuration_path` does not get in initialized. I exposed it by making it a parameter of the `init` method. Then you can pass in different [...]

Like I said above, I want to create two host routing manager on two different IP address(even on two different VLAN), so I think MULTIPLE ROUTING MANAGER can help me, am I on the wrong track? Or can you explain what is MULTIPLE ROUTING MANAGER use for ? Thanks very much.

ManuelKugelmann, 2020-03-17:
I'm no expert concerning this use case, but I think this could be the right way.

Extend the init() method of the application implementation to take configuration_path as a parameter:
`bool application_impl::init(const std::string& configuration_path) ...` to pass different config files/folders to your applications (you will probably have one for each routing manager including all the apps that use that routing manager). You can put the overlay configuration into the applications section of the config `https://github.com/GENIVI/vsomeip/blob/master/documentation/vsomeipUserGuide`.

But if you have two separate configs, one for each routing manager, why do you need overlays?

xuguoxuan, 2020-03-18:
> I'm no expert concerning this use case, but I think this could be the right way.
>
> Extend the init() method of the application implementation to take configuration_path as a parameter:
> `bool [...]

Since I want to own these two application in one same proccess program , so I need overlays. I will extend the init() method like you suggested above, and I'll tell you my result later. Thank you anyway.

lutzbichler, 2020-03-24:
The intended feature for this use case are configuration overlays. Just assign the overlay within the application section of the main configuration and overwrite the necessary parameters (at least unicast & network).

Shangwei1024, 2020-03-31:
I have the same problem,can you give me your email to communicate with the problem?

xuguoxuan, 2020-04-02:
> I have the same problem,can you give me your email to communicate with the problem?
> 
> my email : [ysw201302@163.com](mailto:ysw201302@163.com)

my email 1143239746@qq.com, you can communicate me that doesn't matter,
 but I think that sharing our experience of vsomeip here may be better :-D

closed 2020-06-02 by lutzbichler

zhengzhe2, 2021-12-10:
> > I have the same problem,can you give me your email to communicate with the problem?
> > my email : [ysw201302@163.com](mailto:ysw201302@163.com)
> 
> my email [1143239746@qq.com](mailto:1143239746@qq.com), you can communicate me that doesn't matter, but I think that sharing our experience of vsomeip here may be better :-D
我也有同样的问题，最后你是怎么解决的？麻烦回复一下，谢谢