COVESA/vsomeip #115: rtti not support on aosp

chengyi, 2020-03-05:
very glad to see the release of vsomeip3, but the android library like liblog is built with no rtti,
then android system library's abi is not compatiable with vsomeip library, so they can't be linked together, must I re-compile full aosp code with rtti?

and boost library must be built using ndk which has diffrent symbol namespace with aosp's c++ standard library(e.g. ndk's `string` has `std::__ndk1::string`, while aosp's `string` has `std::string`), how to build boost using aosp tree?

jayhou, 2020-04-08:
You can refer this article for aosp compile boost libs .  https://www.jianshu.com/p/51008521c490 ^_^.    
I have not found there is compatiable problem with rtti lib link to no rtti liblog libs.

chengyi, 2020-04-12:
@jayhou zhoujielun is coming haha:-)
thanks jielun.

closed 2020-04-12 by chengyi

jayhou, 2020-04-13:
@chengyi 哈哈！